robotology/icub-main #402: How to stabilize gaze controller?

frank-foerster, 2016-12-20:
Hi,

I have been "playing around" with the gaze controller in conjunction with the simulator in the last few days using the Python bindings and found it to become unstable in certain conditions.
So I was wondering what the likelihood was of this occurring on the real robot and, even if unlikely on the real robot, how to fix it for the simulator. 

Maybe there are some rules of thumb and/or parameters that help stabilizing it?

The only issue that sounded related to the one I encounter is on the the old rc-hackers mailing list [here](https://sourceforge.net/p/robotcub/mailman/message/31795583/). But opposite to this issue I'm not constantly steaming new target coordinates to the gaze controller - I only ask it three times to change the gaze.
I tried pretty much everything in the controller interface that "sounded like" as if it could render it more stable, but it still seems to be quite temperamental and becomes unstable if I don't call the function `IGazeControl::stopControl()` fast enough after issuing the last `IGazeControl::lookAtFixationPoint()`.
And even when calling it "at the right time", it still sometimes becomes unstable (in 1 out of 10 trials).

The script I am using is too long in its entirety to post, but in terms of motor control if has the following properties:

- I use position control on the left and right arms to get them into a kind of home position
- Subsequently I use the cartesian interface (ICartesianControl) for the left arm to get the robot's hand to some target coordinate. I quasi-simultaneously issue a command to IGazeControl - thus, upper body and head/eyes move at the same time while being controlled by two different controllers.
- I only issue three `IGazeControl::lookAtFixationPoint()` commands related to a starting and a target point/coordinate on the table: look at start coordinate, a little later: look at target coordinate, later again: look (back) at start coordinate (in the videos below you see messages saying so in the command line which indicate when precisely these commands are issued).
- In order to prevent the head from going crazy, I call the `stopControl()` function 1 second after the last call of `lookAtFixationPoint()` in order to give the controller time to move the head/eyes, but at the same time not to have it go crazy
- In addition to calling `stopControl()` at the "right time" I also do:
  + set the neck angle tolerance to 5 degrees using `setNeckAngleUserTolerance(deg)`. This is done as the motion otherwise appears to take forever to complete
  + activate stabilization using `setStabilizationMode(True)`
  + limit neck roll to values in [-20,20] using `bindNeckRoll(min, max)`
- I'm using the standard config file for iKinGazeCtrl (--from configSim.ini)

Instead of calling `stopControl()` I also tried to deactivate tracking after the last call of `lookAtFixationPoint()` but this is far less efficacious in stabilizing the head movements - only in 3 out of 10 the head does not go wild after issuing the latter command. 

Here are four videos of the combined reaching / gazing behaviour.
In the first two I use `stopControl()` for stabilization (see also command line output), on the last two
I out-commented the call just to show how much worse it gets if `stopControl|()` is not issued.

[Video 1 (using stopControl())](https://vimeo.com/196441728): I issue my script twice. The first time around the controller "catches itself" at the end, the second time around the head goes haywire. 

[Video 2 (using stopControl())](https://vimeo.com/196442127): same as in Video 1, the head goes a slightly less crazy in the second run of the script

[Video 3 (NOT using stopControl())](https://vimeo.com/196442274): Already in the first run of the script the head spins out of control

[Video 4 (NOT using stopControl())](https://vimeo.com/196442396): Same as in video 3: the head gets out of control very fast

Any ideas what is causing this and how I could fix it?

But most importantly: Is it safe to run my script on the real robot? The "crazy head movements" in the simulator are probably not achievable by real motors. But if they would, it would probably tear its own head of.

pattacini, 2016-12-20:
Hi @frank-foerster, 

All the scary and daunting effects you've seen have been happening exclusively on `iCub_SIM`. The gaze controller runs reliably on the real robot in the large number of conditions we managed to test throughout the past years.

Could you launch `iKinGazeCtrl` with the option `--imu::switch off` and run again your experiments to see whether disabling the IMU stabilization helps? If you'll get an improvement we can make this as default value for the simulator.

I do expect an improvement, which might be not decisive though. Unfortunately, it very depends on how the current simulator is tuned and I'm afraid we cannot plan an effort to sort out this misbehavior given that many resources are already allocated on Gazebo.

pattacini, 2016-12-20:
Sorry the correct option is `--imu::mode off`

frank-foerster, 2016-12-20:
Hi @pattacini ,
thanks for the hint to switching off IMU. I'll test it later or tomorrow morning and will report back. 
If that doesn't help I'll probably have to give the gazebo version another try.

Otherwise thanks for the reassurance that I won't see this on the real robot - just transitioning from simulator to the real thing. 

I switched back from the Gazebo version to the old simulator as ARE was not working at all on Gazebo a few weeks ago. But I saw there has been a bunch of activity there, so I assume it's time to try the Gazebo version again.

pattacini, 2016-12-20:
My suggestion is to isolate a very minimal C++ snippet that is causing instability on the simulator: it might be an excerpt of your experiment. Thus, I could easily review it, just to make sure there aren't incongruities to then run it on the simulator and on the robot as well.

PS: I'm not the right person to ask for running Python code 😄

frank-foerster, 2016-12-21:
So I finally got around to re-run my script a few times with the `--imu::mode off` setting of `iKinGazeCtrl` and now everything works like a charm. 
I ran it 10 times with calling `stopControl()` and 10 times without and the head never became instable. So it looks like this should be the default setting for the old simulator.

One last question: Does the function `setStabilizationMode(bool s)` now do anything at all given that IMU is deactivated from the outset? 
Should I remove the call to it?

Many thanks for your help.

P.S: I think the SWIG-generated Python interface is really just a wrapper which internally calls
the (compiled) C++ code or something very similar to it, at least that's my understanding of it. So for the problems like the one I had here it probably doesn't make a difference at all whether I interface with the gaze controller via some C++ client or some Python interface.

pattacini, 2016-12-21:
Thanks @frank-foerster for testing: I've just applied the change of the default value in `icub-main/master` and reported it in `devel` as well. Happy to know that the simulator behaves properly now.

Coming to your questions:

1. Requests to `setStabilizationMode()` won't have any effect, provided that `imu::mode` is `off`. The output value will be thus `false`.
2. Of course, the gaze controller will be triggered again at the subsequent calls to any `look`-like methods and will initiate another movement.

Yeah, I'm aware of the bindings, and in this precise case Python snippets wouldn't have made any difference with respect to their C++ counterparts. I was just kidding a bit about the fact that I'm a Python hater 😉

frank-foerster, 2016-12-22:
Thanks for the clarifications.

closed 2016-12-22 by frank-foerster

ZiaMohiUDin, 2019-09-29:
@frank-forester
Can you please guide me how can I control icub with python..