robotology/icub-main #403: Logger reports FT sensors saturation after 10sec

S-Dafarra, 2016-12-20:
We noticed a "huge" delay between the first occurrence of the FT sensors saturation and the relative error on the logger, around 10 sec instead of 1 sec, as it is supposed to be looking at the message  
``
273	160,029882	ERROR	 from BOARD 10.0.1.8 (EB8), src CAN2, adr 13, time 9248s 897m 531u: (code 0x01000002, par16 0x0004 par64 0x0000023600000000) -> HW: strain values saturation. In p16 there is the channel involved, in p64 there are the saturation counts for the ``**last sec**``of both types (lower: LS32bits, upper MS32bits + 
``.

Even if the counter seems working correctly, it is difficult to debug eventual high level control failures with such an high delay (for example we cannot know whether the saturation occurred continuously or not, during those 10sec).

@traversaro @DanielePucci @francesco-romano @gabrielenava  @marcoaccame @valegagge

traversaro, 2016-12-20:
This used to work fine. We suspect there have been some regression either in `embObjAnalogSensor` or in the firmware.

iron76, 2016-12-21:
😮

marcoaccame, 2016-12-21:
The behavior has not changed and is correct. The diagnostics message is just misleading. 

It should be:

```
HW: strain values saturation. In p16 there is the channel involved, in p64 there are the 
saturation counts for the last 1000 readings of both types (lower: LS32bits, upper MS32bits
```

What I think is that the rate of the STRAIN is 10 ms, hence 10 ms * 1000 = 10 seconds.

marcoaccame, 2016-12-21:
Now, discussion is what we want the diagnostics to behave: 
1. every 1000 times is happens (as now)
2. every second if it happens,
3. any other proposal

S-Dafarra, 2016-12-22:
I would chose the second option. Since we will use it to debug high level control failures, having this information with a second delay at maximum, it will be much more effective.

traversaro, 2016-12-22:
> The behavior has not changed and is correct. The diagnostics message is just misleading. 

For completeness, I clearly remember ( @francesco-romano @DanielePucci please correct me )  that some time ago the message was printed every second. Given that the behavior of the firmware/device did not change, I think that at some point the period of the STRAIN broadcast was modified from 1 ms to 10 ms .

DanielePucci, 2016-12-23:
I sincerely remember 1s too, but I am not be completely sure.

marcoaccame, 2017-01-03:
OK, I will change the FW in devel to emit these diagnostics messages at 1 Hz (if any saturation happens).

DanielePucci, 2017-01-05:
👍

pattacini, 2017-03-28:
@marcoaccame can this issue be considered finally addressed?

marcoaccame, 2017-03-30:
yes. 
since ems.hex v 3.0 of 03feb17 (https://github.com/robotology/icub-firmware-build/commit/92f20750223b76f681efab1158926fb8f5eae94f).

closed 2017-03-30 by marcoaccame