robotology/icub-main #405: Consistency of head version option among different modules

gsaponaro, 2016-12-22:
Hi, following https://github.com/robotology/QA/issues/167 we are now using `iKinGazeCtrl --head_version 2.0` in Lisbon. However, I noticed that `gravityCompensator` and `wholeBodyDynamics` still use the "old" option name `--headV2`.

Perhaps it makes sense to make the option `--head_version x.y` on the two latter modules as well? If you want, I can prepare the corresponding PRs in `icub-main` and `robots-configuration`. If not, feel free to close this issue. :)

pattacini, 2016-12-22:
Hi @gsaponaro, we are aware of this and in fact #285 is not closed yet; thanks for reporting though.

Feel free to contribute with a PR; @randaz81 might consider reviewing it.

closed 2016-12-22 by pattacini

gsaponaro, 2016-12-22:
OK, cool.

Prepared a PR for review: https://github.com/robotology/icub-main/pull/406. If that is merged, then I'll send a corresponding PR in `robots-configuration` too (which will affect only our robot `iCubStartup.xml`).