robotology/icub-main #409: Problems in launching depth2kin

giuliavezzani, 2017-01-13:
I cannot launch **depth2kin** tool on the `iCubGenova01`. 
The error I got is the following:

```
||| did not find config.ini
yarp: cannot read from config.ini
[INFO]loading experts from file: /home/icub/.local/share/yarp/contexts/depth2kin/calibration_data.ini
depth2kin: /usr/include/eigen3/Eigen/src/Core/DenseBase.h:261: void Eigen::DenseBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = Eigen::Map<Eigen::Matrix<double, -1, -1, 1> >; Eigen::Index = long int]: Assertion `rows == this->rows() && cols == this->cols() && "DenseBase::resize() does not actually allow to resize."' failed.
Aborted (core dumped)
```

Any suggestions?

pattacini, 2017-01-13:
@giuliavezzani it might be that `yarp/devel` needs to be recompiled. Please try again when you have time a report here the outcome.

giuliavezzani, 2017-01-13:
I updated and recompiled `yarp/devel` and `icub-main/devel` both on the `icub21` and the `pc104`, but I still got the same output as before.

giuliavezzani, 2017-01-13:
Thanks a lot!! :smile:

pattacini, 2017-01-14:
Related issue: https://github.com/robotology/yarp/issues/1062.