robotology/icub-main #417: Inconsistent to strange ARE behaviour

frank-foerster, 2017-01-26:
Hi,

I have been testing the ActionsRenderingEngine on our physical iCub lately but get very inconsistent behaviour out of it. 

So I'm wondering if I'm doing anything wrong.

I took two videos of what I think is inconsistent / strange behaviour to illustrate the point.

In the first video I have it execute two identical `touch` commands via rpc'ing into `/actionsRenderingEngine/cmd:io`. Despite the two touch commands being exactly the same, the ensuing behaviour is radically different. The first execution is roughly what I would have imagined in that the iCub moves the hand to a position that approximately corresponds to the specified coordinate, that is, where the cartesian controller, when used on its own without ARE, would take the arm+hand.

The second touch command leads to its arm being taken upwards instead of toward the target position. Also the hand orientation is very different.

command: `touch (-0.52 0.15 0)`
[Video double-touch](https://vimeo.com/201103887)

I tried this for several rounds and approxmately every 2nd touch command works as I would imagine it, though there also seems to be some strange limit on the distance (x-coordinate) such that I never managed to have it actually touch the object, despite it being in reaching distance. I increased/decreased x from -0.52 up to -0.65  but it never reaches those 13cm further. 

In the second video I issue a `grasp` followed by a `drop` command. target coordinates are identical to the ones used in the `touch` example above. The precise commands -  all entered via the  rpc /cmd:io port - are:

`grasp (-0.52 0.15 0.03 0 1 0 3.142)`

and

`drop over (-0.52 0.15 0.03)`

[Video: grasp and drop](https://vimeo.com/201104828)

As you can see in the video the grasp does not succeed, its hand is once again far too high for the target object, but I had instances where it came somewhat close the object with pretty much the same coordinates - I typically varied the height (z coord) by a few centimeters. 

More shockingly, the drop command leads to it going with its hand through - what would be - the table surface and not even close to any of the other positions it had previously gone to.

As I have never seen how the behaviour should look like when ARE gets the respective commands, I'm not sure what the "gold standard" would be, but I'm pretty sure that what I am seeing is not the intended behaviour. 

So here are some details and questions:

- I deliberately do not use vision for the time being and assume that the vision systems do not influence command execution when absolute coordinates are specified as above. Is that correct?

- Is there any calibration I should perform prior to executing above commands? I'm asking because I noticed that the iCub will only close its hand during the grasp command when I run a `calibration hand` command beforehand. If I don't do that the arm movement is being produced but the hand never actually closes. 

- I'm starting ARE via yarpmanager using [this script](https://github.com/robotology/icub-main/blob/master/app/actionsRenderingEngine/scripts/actionsRenderingEngine.xml.template) from the repository, but I only start ARE, none of the other modules.
Prior to starting ARE I start up the modules specified in [iCubStartup.xml](https://github.com/robotology/icub-main/blob/master/app/iCubStartup/scripts/iCubStartup.xml.template) minus the kinematic solvers for the legs and minus the fingersTuner (is that one new?). 
I added the parameter `--context actionsRenderingEngine` to make sure it picks up the correct config-file.

- I used the cartesian controller (ICartesianControl) prior to using ARE and that one seems to work pretty much fine on its own. In the examples above I'm exclusively running ARE though, that is, there is not other competing control process running in the background.

- In terms of configuration files I'm using the [standard ARE config file](https://github.com/robotology/icub-main/blob/master/app/actionsRenderingEngine/conf/config.ini), the only difference being that I set `part_used` to `right_arm`.

- I did not run a `calibration table` command prior to executing the other commands due to my "paper table", but I modified the table height in `kinematic_offsets.ini` to -0.06 when debugging as I thought that might be the issue (was -0.01x) (iCub refusing to reach the target because it's underneath the table surface).

I have the log-files here and can upload them if that helps, but hey contain a few more command executions than the one I show in the videos. I haven't seen any error messages in the log-file generated by ARE.

So any suggestions / speculation on what is going wrong here are welcome.

vtikha, 2017-01-26:
@frank-foerster Thank for the videos and details, it made things clear. The issue that you are noticing seems to be due to the detection of external forces from the force-torque sensor of the arm in use. By sending commands just through `cmd:io` the force-torque sensor does not reset and drifts. Therefore during an action it detect some false spikes and stops from completing the action. This is needed for safety when calibrating on the table or when stopped by the user etc. I would suggest to do the following things:

- before any action, send `0` to wholeBodyDynamics to recalibrate the sensors. `yarp rpc /wholeBodyDynamics/rpc:i` and enter `0`. Please note that the robot needs to be completely still when sending this command. We typically send this command every time before requesting an action. Obviously this is done by means of a `manager module` and should not be done by `ARE` itself.

- in the ARE config file `config.ini` that you provided, please increase the `external_forces_thresh` on the arm. Currently it is at `3` I would recommend to put it initially to `5`. 

There are no errors in the `ARE` log-file but you can see the external force threshold being exceeded, I don't remember the exact terminology but if you upload the log-file I can tell you where it is written. 

Let me know if this solves the problem for you, hopefully it will 🤞

frank-foerster, 2017-01-26:
Hi @vtikha ,

thanks for the fast reply. I'll try resetting wholeBodyDynamics in the way you described and will increase the threshold parameter, and will report the results here. 

I just had another look through the log file of the actionsRenderingEngine, and I guess the message regarding external forces is this one:

`<WARNING> *** actionsRenderingEngine/right_arm: contact detected on arm: external force [-0.905 -1.267 -2.816], (3.21826>3) => stopping arm`

I also regularly encounter this one:
`<INFO> *** actionsRenderingEngine/right_arm: timeout (4) expired while reaching`

_after_ lines that seem to document the performance of reaching behaviour: 

`<UNDEFINED> *** actionsRenderingEngine/right_arm: reaching... xdhat=[-0.280  0.233  0.197] |e|=0.018 [m]; odhat=[-0.028 -0.885  0.465  2.988] |e|=0.023`

Not sure if that's another problem (too tight time thresholds?).

But more later on.

Thanks again.

vtikha, 2017-01-26:
Hi @frank-foerster The `warning` message proves the point. I believe setting the `external_forces_thresh` to 5 will correct this. As for the other ones, they are probably due to the fact that you are using `ARE` in `impedance` mode and the controller does not manage to reach the required position in time. `ARE` is configured to have a `reach_tol` of 2 cm (ARE config file). I would not change that but instead set `impedance` to `off`.

frank-foerster, 2017-01-26:
Hi @vtikha,
so I just applied the two adjustment - changing `external_forces_thres` to 5 as well as having wholeBodyDynamics recalibrate before each command - and the command execution looks much better now. 
Thanks for the hint towards the `impedance` setting - I might try that. 
If you don't mind, I'd have two more questions:

1. `drop over (some_coordinates)` still leads to its arm going through my paper table, even though I'm using the very same coordinates as for the `grasp` and `touch` commands. Adding the params `above gentle` doesn't change that. 
If I don't specify any  coordinates though, that is `drop above gentle`, it looks much more reasonable, though I'd prefer to specify the location. Is there a way to have it drop the object over a specific point without it going underneath it? 

2. At the end of the day I'd like to use ARE in combination with cartesian control of the arms. So I'd like to control the arm movements via cartesian control and just do an occasional grasp and drop without having ARE move the arm more than absolutely required. 
So I was wondering if I can run a module that instantiates cartesianControl concurrently with ARE without the two getting in each others way - given they're controlling the same joints. I was thinking about sequencing the control commands, call `ICartesianControl_instance.stopControl()` etc. prior to issuing commands to ARE.

Or would it in this case be advisable to use the actionPrimitives layer directly without running ARE.

Many thanks again.

pattacini, 2017-01-26:
Hi @frank-foerster 

#### Dropping
- `above` doesn't seem to be a recognized option for dropping.
- The option `gently` (not `gentle`) makes use of the table's height for the z coordinate, hence you are supposed to have carried out a `calib table` command beforehand. Anyway, `gently` is [_mutual exclusive_][1] with `over`, apparently.
- At any rate, I've just tested with the simulator and `drop over (x y z) [arm]` does work, reaching the point `(x y z)` as issued. Consider that `touch` and `grasp` have proper (internal offset) wrt the specified target. You might want to check that by inspecting the `ARE` printouts. Therefore, when you yield a `drop` you shouldn't pass the exact same coordinates of `touch` and `grasp`.

#### ARE combined with other Cartesian Clients
It is definitely possible to go for this combination. Just bear in mind that `ARE` and any external client must not send commands simultaneously - of course - and that both of them have to take care of their respective [**Cartesian contexts**][2].

[1]: ../blob/master/src/modules/actionsRenderingEngine/src/MotorThread.cpp#L2223-L2225
[2]: http://wiki.icub.org/iCub/main/dox/html/icub_cartesian_interface.html#sec_cart_contextswitch

frank-foerster, 2017-01-27:
Hi @pattacini ,

Thanks for the quick reply. 

If `above` is not an option, then the [documentation](http://wiki.icub.org/iCub_documentation/group__actionsRenderingEngine.html) needs updating: 

[in the context of using _drop_ without target coordinate] "Optional parameter "side" or "above" can be provided in order to select the hand orientation that the robot should try to mantain during the drop action"
[in the context of using _drop_ with target coord.] "Also in this case the optional parameter "side" or "above" can be specified"

(Btw: I'm happy of adding to the documenation if somebody give me write access, the information of sending a 0 to wholeBodydynamics prior to issuing each command seems crucial to operate ARE at all in any meaningful manner, but see below)

I'll retest the drop command it in a minute without `above` option, not sure if I misspelled `gently` here, or during the actual command execution, and have a closer look at the log files.

**Regarding the _cartesian contexts_**: 
Would I need to save the context of my cartesian controller prior to sending a command to ARE and restore it before using it again? 

My intuition was that I wouldn't need to do that because I'm not actually changing the context of that very (local) controller. That is, I'm not activating or deactivating joints, change limits, etc.. 
My approach would have been to  just issue a <cartCtrl_instance>.[stopControl()](http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#a2118799f76bef79343ccdefa8d551a9a) prior to sending a command to ARE,  and once ARE is finished executing it, send an _interrupt_ command to /actionsRenderingEngine/rpc, prior to reissuing commands via the cartesian controller. 

Unrelated to the above: 
Yesterday I went through the ARE source code and an odd thing I discovered is, that - if the code does what it looks like it's ought to do - ARE is already issuing a recalibration command to _wholeBodyDynamics_ in the case of __grasp (...)__ being issued (line 1689 of [MotorThread.cpp](https://github.com/robotology/icub-main/blob/master/src/modules/actionsRenderingEngine/src/MotorThread.cpp)), prior to actually having the arm move. 
The fact that me sending this command manually makes the world of a difference, indicates that either _wdbRecalibration()_ (line 267 of [MotorThread.h](https://github.com/robotology/icub-main/blob/master/src/modules/actionsRenderingEngine/include/iCub/MotorThread.h) is not working properly, that my yarpmanager is misleading me when it claims that the relevant rpc connection between ARE and WBD had been established, or that for some reason the RPC call leads to different function.

Anyways, I will re-test, re-investigate, and report the results here.

frank-foerster, 2017-01-27:
After having performed a few more tests I found that a save coordinate for dropping an object appears to be approximately 30 to 40 cms above the original coordinate of the object, in the sense that the arm won't hit the table when dropping it.

So, as example, if the original object is at (-0.52 0.15 0.01)

`grasp (-0.52 0.15 0.01 0 1 0 3.142)`

a save drop command would be 

`drop over (-0.52 0.15 0.4)`

and it appears to be consistently save, but the x-y location is off by various centimeters.
The latter could have to do with the following though.

I found that I still get some warnings about perceived external forces stopping the arm, indicating that I should maybe increase `external_forces_thres` to 11 in order to avoid the switching off of joints.

Is there some upper limit that should never be exeeded? And are these force measurements in Newton? 

I'm closing this issue for now, as the most important issue - need to recalibrate WBD prior to issuing ARE commands - is solved.

closed 2017-01-27 by frank-foerster

pattacini, 2017-01-27:
Quick comments:

- `drop` documentation updated, thanks for spotting this out.
- `ARE` can reset `WBD` on its own, but you have to connect `/ARE/wbd:rpc` to `/WBD/rpc`.
- Ok for not handling the contexts. No need to send `interrupt` if you wait until `ARE` is done (you'll receive the `ack` via rpc). If you send an `interrupt`, then it's likely you'll have to send a `reinstate` - if I remember correctly.
- Each command has its own offsets that can explain the behavior you saw.
- I would recommend you to use `take` rather than `grasp`.
- Forces are measured in Newton, yep.

frank-foerster, 2017-01-27:
Thanks for the comments. 

I have double-checked today that the connection between `/ARE/wbd:rpc` and `/WBD/rpc` was established and still got inaccurate grasping when not sending a 0 manually to the `/WBD/rpc` port. As soon as I sent it manually prior to issuing an ARE command things improved considerably.
Don't have time to debug this now, but might do once my current experiment is finished. 

I had a go at `take` as well. I'm still testing which of the two better integrates with my current script. 

Thanks for the help again.