robotology/icub-main #419: Compiling icub-main on windows fails

BrutusTT, 2017-01-27:
Hi,

compiling icub-main 1.4.0 and icub-main master as Release fails to compile iCub_SIM on Windows 7 with VS 2013 with the following error message:

MSVCRT.lib(crtexew.obj) : error LNK2019: unresolved external symbol _main referenced in function ___tmainCRTStartup

Found suggestions to create a main method, but its there. Found suggestions to play around with the linker setup but it looks correct and changing them doesn't help.

Any ideas what might cause that?

drdanz, 2017-01-27:
Latest icub-main release is 1.6.0, is there a reason why you want to compile 1.4.0?

Anyway, are you using VS 2013 express edition or enterprise?

Which CMake version are you using?

BrutusTT, 2017-01-31:
No particular reason for using 1.4.0 but the master and 1.6.0 didn't worked either.

VS 2013 Enterprise and CMake 3.6.2

closed 2017-03-28 by pattacini