robotology/icub-main #424: ARE: pfar uses hand mid point instead of index finger when pointing

Tobias-Fischer, 2017-02-15:
Hi @pattacini,
As briefly discussed at the WYSIWYD review meeting, `pfar` of `actionsRenderingEngine` uses the mid point of the hand to calculate where to point at, but as the pointing is actually performed by the index finger there is an offset.
It would be great if the index finger could be used to increase the accuracy of the pointing.

Best,
Tobi

pattacini, 2017-02-17:
Development commenced in [**`fix/are-point-far`**](https://github.com/robotology/icub-main/tree/fix/are-point-far).

pattacini, 2017-02-17:
Hi @Tobias-Fischer 

It should be fixed in https://github.com/robotology/icub-main/commit/ac8ea6f610e13405c3cb2da37e87437e459084e8.

Find here a useful [**`Gist`**][1] you might want to use for testing.

[1]: https://gist.github.com/pattacini/b79751ba8751efde15d076dee76222da

closed 2017-02-17 by pattacini

Tobias-Fischer, 2017-02-17:
Cool, thanks a lot! I'll give it a try next week.