robotology/icub-main #426: Valgrind errors using DragonflyDeviceDriver

Nicogene, 2017-03-03:
Yesterday I found these valgrind errors running the dragonfly on iCub : 

```
==10181== Syscall param ioctl(generic) points to uninitialised byte(s)
==10181==    at 0x5D951C7: ioctl (syscall-template.S:81)
==10181==    by 0xA782044: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA772FEE: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA7730CE: dc1394_camera_enumerate (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA13EA81: CFWCamera_DR2_2::Create(yarp::os::Searchable&) (FirewireCameraDC1394-DR2_2.cpp:327)
==10181==    by 0xA13503D: yarp::dev::DragonflyDeviceDriver2::open(yarp::os::Searchable&) (DragonflyDeviceDriver2.cpp:97)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50CD781: yarp::dev::ServerFrameGrabber::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x509E43A: yarp::dev::Drivers::yarpdev(int, char**) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==  Address 0xffeffe518 is on thread 1's stack
```

```
==10181== Conditional jump or move depends on uninitialised value(s)
==10181==    at 0xA7729E6: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA773045: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA7730CE: dc1394_camera_enumerate (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA13EA81: CFWCamera_DR2_2::Create(yarp::os::Searchable&) (FirewireCameraDC1394-DR2_2.cpp:327)
==10181==    by 0xA13503D: yarp::dev::DragonflyDeviceDriver2::open(yarp::os::Searchable&) (DragonflyDeviceDriver2.cpp:97)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50CD781: yarp::dev::ServerFrameGrabber::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x509E43A: yarp::dev::Drivers::yarpdev(int, char**) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x4009F1: main (in /usr/local/src/robot/yarp/build-pc104/bin/yarpdev)
```

```
==10181== Use of uninitialised value of size 8
==10181==    at 0xA772BAC: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA773045: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA7730CE: dc1394_camera_enumerate (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA13EA81: CFWCamera_DR2_2::Create(yarp::os::Searchable&) (FirewireCameraDC1394-DR2_2.cpp:327)
==10181==    by 0xA13503D: yarp::dev::DragonflyDeviceDriver2::open(yarp::os::Searchable&) (DragonflyDeviceDriver2.cpp:97)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50CD781: yarp::dev::ServerFrameGrabber::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x509E43A: yarp::dev::Drivers::yarpdev(int, char**) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x4009F1: main (in /usr/local/src/robot/yarp/build-pc104/bin/yarpdev)
```

```
==10181== Use of uninitialised value of size 8
==10181==    at 0x5CFA041: _itoa_word (_itoa.c:180)
==10181==    by 0x5CFE249: vfprintf (vfprintf.c:1641)
==10181==    by 0x5DAA394: __vsnprintf_chk (vsnprintf_chk.c:63)
==10181==    by 0xA77FBAE: dc1394_log_debug (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA772D28: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA773045: ??? (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA7730CE: dc1394_camera_enumerate (in /usr/lib/x86_64-linux-gnu/libdc1394.so.22.1.10)
==10181==    by 0xA13EA81: CFWCamera_DR2_2::Create(yarp::os::Searchable&) (FirewireCameraDC1394-DR2_2.cpp:327)
==10181==    by 0xA13503D: yarp::dev::DragonflyDeviceDriver2::open(yarp::os::Searchable&) (DragonflyDeviceDriver2.cpp:97)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50CD781: yarp::dev::ServerFrameGrabber::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6
```

```
==10181== Conditional jump or move depends on uninitialised value(s)
==10181==    at 0x5CFA048: _itoa_word (_itoa.c:180)
[... 137 lines cut ...]
==10181==    by 0x50CD781: yarp::dev::ServerFrameGrabber::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B7112: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x50B81B4: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x509E43A: yarp::dev::Drivers::yarpdev(int, char**) (in /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.69.6)
==10181==    by 0x4009F1: main (in /usr/local/src/robot/yarp/build-pc104/bin/yarpdev)
==10181==  Address 0xffeffe590 is on thread 1's stack
```

```
==10181== HEAP SUMMARY:
==10181==     in use at exit: 1,601,696 bytes in 2 blocks
==10181==   total heap usage: 419,535 allocs, 419,533 frees, 68,437,885 bytes allocated
==10181== 
==10181== LEAK SUMMARY:
==10181==    definitely lost: 1,601,664 bytes in 1 blocks
==10181==    indirectly lost: 0 bytes in 0 blocks
==10181==      possibly lost: 0 bytes in 0 blocks
==10181==    still reachable: 32 bytes in 1 blocks
==10181==         suppressed: 0 bytes in 0 blocks
==10181== Rerun with --leak-check=full to see details of leaked memory
==10181== 
==10181== For counts of detected and suppressed errors, rerun with: -v
==10181== Use --track-origins=yes to see where uninitialised values come from
==10181== ERROR SUMMARY: 758 errors from 102 contexts (suppressed: 0 from 0)
```

Nicogene, 2017-11-02:
Reminder for me and @barbalberto: try to get a pc104 to try to fix this issue.

pattacini, 2023-07-30:
Dragonfly cameras are deprecated.

closed 2023-07-30 by pattacini