robotology/icub-main #427: Valgrind errors using FrameGrabberGui2

Nicogene, 2017-03-06:
When I connect the `frameGrabberGui2` to the port opened by `grabber` (using `usbCamera`, I have to investigate if I have the same error using `dragonfly2`), I get this error:

```
==11285== Thread 14:
==11285== Syscall param write(buf) points to uninitialised byte(s)
==11285==    at 0x5E556FD: ??? (syscall-template.S:84)
==11285==    by 0x6B33FB6: ACE::send_n_i(int, void const*, unsigned long, unsigned long*) (in /usr/lib/libACE-6.3.3.so)
==11285==    by 0x61D7B0F: ACE::send_n(int, void const*, unsigned long, ACE_Time_Value const*, unsigned long*) (ACE.inl:188)
==11285==    by 0x61D7C6A: ACE_SOCK_Stream::send_n(void const*, unsigned long, ACE_Time_Value const*, unsigned long*) const (SOCK_Stream.inl:119)
==11285==    by 0x61D83FA: yarp::os::impl::SocketTwoWayStream::write(yarp::os::Bytes const&) (SocketTwoWayStream.h:144)
==11285==    by 0x61AB36B: yarp::os::impl::BufferedConnectionWriter::write(yarp::os::OutputStream&) (BufferedConnectionWriter.cpp:165)
==11285==    by 0x61AD79D: yarp::os::Carrier::reply(yarp::os::ConnectionState&, yarp::os::SizedWriter&) (Carriers.cpp:290)
==11285==    by 0x6247A55: yarp::os::impl::Protocol::reply(yarp::os::SizedWriter&) (Protocol.h:137)
==11285==    by 0x627E053: yarp::os::impl::StreamConnectionReader::flushWriter() (StreamConnectionReader.cpp:41)
==11285==    by 0x6247AAD: yarp::os::impl::Protocol::endRead() (Protocol.h:159)
==11285==    by 0x6230BEE: yarp::os::impl::PortCoreInputUnit::run() (PortCoreInputUnit.cpp:366)
==11285==    by 0x6283AD2: theExecutiveBranch(void*) (ThreadImpl.cpp:104)
==11285==  Address 0x9eb9494 is 36 bytes inside a block of size 1,024 alloc'd
==11285==    at 0x4C2E80F: operator new[](unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==11285==    by 0x61DC0ED: yarp::os::ManagedBytes::ManagedBytes(unsigned long) (ManagedBytes.cpp:24)
==11285==    by 0x61AAD33: yarp::os::impl::BufferedConnectionWriter::addPool(yarp::os::Bytes const&) (BufferedConnectionWriter.cpp:88)
==11285==    by 0x61AAF1F: yarp::os::impl::BufferedConnectionWriter::push(yarp::os::Bytes const&, bool) (BufferedConnectionWriter.cpp:115)
==11285==    by 0x61AB6F9: yarp::os::impl::BufferedConnectionWriter::appendBlockCopy(yarp::os::Bytes const&) (BufferedConnectionWriter.h:214)
==11285==    by 0x61ABC76: yarp::os::impl::BufferedConnectionWriter::appendBlock(char const*, unsigned long) (BufferedConnectionWriter.h:356)
==11285==    by 0x61A247E: yarp::os::impl::BottleImpl::write(yarp::os::ConnectionWriter&) (BottleImpl.cpp:534)
==11285==    by 0x619ED3B: yarp::os::Bottle::write(yarp::os::ConnectionWriter&) (Bottle.cpp:167)
==11285==    by 0x50C5EDD: yarp::dev::DeviceResponder::read(yarp::os::ConnectionReader&) (DeviceDriver.cpp:79)
==11285==    by 0x6239150: yarp::os::impl::PortReaderBufferBase::read(yarp::os::ConnectionReader&) (PortReaderBuffer.cpp:439)
==11285==    by 0x622D173: yarp::os::impl::PortCoreAdapter::read(yarp::os::ConnectionReader&) (PortCoreAdapter.cpp:120)
==11285==    by 0x621FF45: yarp::os::impl::PortCore::readBlock(yarp::os::ConnectionReader&, void*, yarp::os::OutputStream*) (PortCore.cpp:1230)
==11285=
```

pattacini, 2017-03-28:
Hi @Nicogene 
What's the status of this issue?

Nicogene, 2017-03-29:
Hi @pattacini,

if nobody fixed it, I'm afraid that this error is still there :pensive:.

In this month I'm busy on `rfsmTools`, after Easter I will try to fix it :smile:

pattacini, 2017-03-29:
Yeah, thanks @Nicogene for being willing to take on this bug fix 😃 

What I'd like to avoid is to just report a problem without taking the countermeasure to sort it out in a reasonable amount of time.

Nicogene, 2017-04-13:
The Valgrind error of this issue disappeared, probably fixed by one of the last commit on `yarp` `devel`.
But Another valgrind error popped out, this one fixed by #438.
Once merged that PR I think we can close this issue.

closed 2017-04-13 by Nicogene