robotology/icub-main #428: Fast-motions with icub's arm

khoramshahi, 2017-03-07:
Hi, 

I would like to generate very fast motion with the new icub's arm (especially the shoulder joint 0 for pitch), and I have two questions regarding this that I would be thankful if you could help me with.
1) how fast can one move this joint (shoulder 0) with regard to safety issues and hardware damage?
2) what is best mode to control for this? Is velocity control with a pre-made velocity profile with a bell-shape a good way to tackle this?

Best,
Mahdi form EPFL

pattacini, 2017-03-07:
Hi @khoramshahi 

I deem your question of quite general interest for the community, hence let me move it to our Q&A system.

pattacini, 2017-03-07:
Issue moved to [robotology/QA #194](https://github.com/robotology/QA/issues/194) via [**ZenHub**](https://www.zenhub.com/)

closed 2017-03-07 by pattacini