robotology/icub-main #429: http://wiki.icub.org/wiki/Downloads is outdated

traversaro, 2017-03-11:
It is only referring to the release 1.4.0 of iCub, but the release 1.6.0 is out since November.

drdanz, 2017-05-05:
Should we add this page to the release instructions or is it there already?
CC @mbrunettini

traversaro, 2017-05-05:
Already added in https://github.com/robotology/icub-main/pull/351 .

traversaro, 2017-06-23:
The issue is actually still there for iCub 1.8.0 : http://wiki.icub.org/wiki/Downloads . 
cc @drdanz @mbrunettini @pattacini

mbrunettini, 2017-06-23:
@drdanz made some changes in the doxigen yesterday.

traversaro, 2017-07-04:
@drdanz Any news? 
In the meanwhile I updated the http://wiki.icub.org/wiki/Downloads page.

aerydna, 2017-11-02:
can we close this issue or is still valid?
@traversaro @drdanz

traversaro, 2018-03-14:
After the release 1.10.0, the Wiki page is outdated again, see http://wiki.icub.org/wiki/Downloads .
cc @drdanz @lucaTagliapietra

mbrunettini, 2018-03-15:
page fixed

traversaro, 2018-10-28:
The page is now outdate after the release 1.12.0, see http://wiki.icub.org/wiki/Downloads .

mbrunettini, 2018-10-29:
links updates

closed 2018-10-29 by mbrunettini