robotology/icub-main #430: problem installing icus-main

azadm, 2017-03-14:
Here is the error that I get after running ``make`` command:

````
dynamic200-113:build azadm$ make
[  0%] Copying manager.py to /Users/azadm/Applications/icub-main/build/bin/.
[  0%] Built target python-scripts
[  0%] Copying icub-cluster scripts to /Users/azadm/Applications/icub-main/build/bin/.
[  0%] Built target cluster-scripts
[  0%] Built target iCubDev
[  3%] Built target ctrlLib
[  5%] Built target skinDynLib
[  7%] Built target iKin
[  8%] Built target iDyn
[ 13%] Built target learningMachine
[ 15%] Built target perceptiveModels
[ 16%] Built target actionPrimitives
[ 16%] Built target icubmod
[ 17%] Linking CXX shared module ../../../../lib/iCub/cartesiancontrollerclient.so
Undefined symbols for architecture x86_64:
  "yarp::sig::Vector::allocGslData()", referenced from:
      yarp::sig::Vector::Vector(unsigned long) in libctrlLib.a(math.cpp.o)
  "yarp::sig::Vector::updateGslData()", referenced from:
      yarp::sig::Vector::Vector(unsigned long) in libctrlLib.a(math.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/iCub/cartesiancontrollerclient.so] Error 1
make[1]: *** [src/libraries/icubmod/cartesianController/CMakeFiles/cartesiancontrollerclient.dir/all] Error 2
make: *** [all] Error 2
````

I am using OS Sierra 10.12.3

claudiofantacci, 2017-03-15:
Hi @azadm first thing first: did you manage to compile YARP from source?
It seems to me (see [YARP issue #1130](https://github.com/robotology/yarp/issues/1130)) that you are experiencing problems.

Anyway, could you please run the following on your terminal and print the output?
```
$ brew info gsl
```
You should get something like this:
```
gsl: stable 2.3 (bottled)
Numerical library for C and C++
https://www.gnu.org/software/gsl/
/usr/local/Cellar/gsl/2.3 (273 files, 8.6MB) *
  Poured from bottle on 2017-01-15 at 19:28:21
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/gsl.rb
```

Please also check this iCub installation guide for further information: http://wiki.icub.org/wiki/Homebrew_Mavericks

azadm, 2017-03-15:
This is what I have:
````
dynamic200-113:~ azadm$ brew info gsl
gsl: stable 2.3 (bottled)
Numerical library for C and C++
https://www.gnu.org/software/gsl/
/usr/local/Cellar/gsl/1.16 (247 files, 8.0MB) *
  Poured from bottle on 2014-10-21 at 23:23:16
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/gsl.rb
````

The thing is that I had both ``yarp`` and ``icub-main`` working on my computer before upgrading to Sierra. They didnt work on the new version of OS so I tried to reinstall everything and now I have got problem with installing ``yarp`` from source and also installing ``icub-main``.

claudiofantacci, 2017-03-15:
This is really strange. I'm on Sierra as well and experience no issues.

Did you delete the build folder and try again everything from scratch?

azadm, 2017-03-15:
yes I did. I even deleted the whole folder and clone it again!

claudiofantacci, 2017-03-15:
It seems that your GSL version is quite old.

Unless you don't have any specific requirement, you should perform a `brew` upgrade.
```
$ brew update && brew upgrade
```
If you want to just update GSL do
```
$ brew update && brew upgrade gsl
```

azadm, 2017-03-15:
I did it:
````
dynamic200-113:build azadm$ brew info gsl
gsl: stable 2.3 (bottled)
Numerical library for C and C++
https://www.gnu.org/software/gsl/
/usr/local/Cellar/gsl/1.16 (247 files, 8.0MB)
  Poured from bottle on 2014-10-21 at 23:23:16
/usr/local/Cellar/gsl/2.3 (273 files, 8.6MB) *
  Poured from bottle on 2017-03-15 at 15:20:52
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/gsl.rb
````
But still the same problem...

claudiofantacci, 2017-03-15:
On which branch YARP and iCub are?

claudiofantacci, 2017-03-19:
Are there any updates on this issue?

azadm, 2017-03-20:
I managed to install both ``yarp`` and ``icub-main`` from source. I don't know why but I had to remove everything and install from scratch...maybe due to the upgrade to Sierra.

closed 2017-03-20 by claudiofantacci