robotology/icub-main #445: yarpmotorgui - error while getting current range

drdanz, 2017-05-05:
@amaroyo commented on [Thu May 04 2017](https://github.com/robotology/yarp/issues/1196)

Hi @randaz81, 

I am not sure whether this has something to do with the [recently updated PWM parameters ](https://github.com/robotology/icub-support/issues/451) or it's just a coincidence, but I realized the following error when I launch yarpmotorgui. All the joints end up with the following error:

[```[ERROR]Error while getting current range, part ...```](https://github.com/robotology/yarp/blob/devel/src/yarpmotorgui/partitem.cpp#L180) It corresponds to a recent change in the devel branch (follow the link)

Anyway, the interface starts and I am able to move all the joints of the robot. Find the full trace below:

```
$ yarpmotorgui 
Warning: QT_DEVICE_PIXEL_RATIO is deprecated. Instead use: 
    QT_AUTO_SCREEN_SCALE_FACTOR to enable platform plugin controlled per-screen factors. 
    QT_SCREEN_SCALE_FACTORS to set per-screen factors. 
    QT_SCALE_FACTOR to set the application global scale factor.
||| configuring
||| default config file specified as yarpmotorgui.ini
||| checking [/Users/Aleks/yarpmotorgui.ini] (pwd)
||| checking [/Users/Aleks/Library/Application Support/yarp/config/yarpmotorgui.ini] (YARP_CONFIG_HOME)
||| checking [/Users/Aleks/Library/Application Support/yarp/yarpmotorgui.ini] (YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/Users/Aleks/IIT/robot/install/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/Users/Aleks/IIT/robot/install/share/iCub/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/Users/Aleks/IIT/robot/install/share/ICUBcontrib/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/Users/Aleks/IIT/robot/src/yarp/build/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/Users/Aleks/IIT/robot/src/icub-main/build/share/iCub/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| did not find yarpmotorgui.ini
yarp: cannot read from yarpmotorgui.ini
QObject::connect: No such signal MainWindow::internalClose()
QObject::connect:  (sender name:   'MainWindow')
QObject::connect:  (receiver name: 'MainWindow')
[WARNING]robotDescriptionServer not found, robot parts will be set manually. 
[DEBUG]Appending /icub/head
[DEBUG]Appending /icub/torso
[DEBUG]Appending /icub/left_arm
[DEBUG]Appending /icub/right_arm
[DEBUG]Appending /icub/left_leg
[DEBUG]Appending /icub/right_leg
[DEBUG]Checking the existence of: /yarpmotorgui0//icub/head/rpc:o 
[DEBUG]ADDRESS is:  
yarp: Port /yarpmotorgui0//icub/head/rpc:o active at tcp://10.0.0.134:10126/
yarp: Port /yarpmotorgui0//icub/head/command:o active at tcp://10.0.0.134:10127/
yarp: Port /yarpmotorgui0//icub/head/stateExt:i active at tcp://10.0.0.134:10128/
yarp: Sending output from /yarpmotorgui0//icub/head/rpc:o to /icub/head/rpc:i using tcp
yarp: Sending output from /yarpmotorgui0//icub/head/command:o to /icub/head/command:i using udp
yarp: Receiving input from /icub/head/stateExt:o to /yarpmotorgui0//icub/head/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[DEBUG]Setting a valid finder 
yarp: Port /yarpmotorgui/head/sequence:o active at tcp://10.0.0.134:10129/
[DEBUG]Initializing interfaces...
[DEBUG]Opening interfaces...
[INFO]head iencs->getEncoders() ok!
[ERROR]Error while getting current range, part  head  joint  0 
[ERROR]Error while getting current range, part  head  joint  1 
[ERROR]Error while getting current range, part  head  joint  2 
[ERROR]Error while getting current range, part  head  joint  3 
[ERROR]Error while getting current range, part  head  joint  4 
[ERROR]Error while getting current range, part  head  joint  5 
[DEBUG]Checking the existence of: /yarpmotorgui0//icub/left_arm/rpc:o 
[DEBUG]ADDRESS is:  
yarp: Port /yarpmotorgui0//icub/left_arm/rpc:o active at tcp://10.0.0.134:10130/
yarp: Port /yarpmotorgui0//icub/left_arm/command:o active at tcp://10.0.0.134:10131/
yarp: Port /yarpmotorgui0//icub/left_arm/stateExt:i active at tcp://10.0.0.134:10132/
yarp: Sending output from /yarpmotorgui0//icub/left_arm/rpc:o to /icub/left_arm/rpc:i using tcp
yarp: Sending output from /yarpmotorgui0//icub/left_arm/command:o to /icub/left_arm/command:i using udp
yarp: Receiving input from /icub/left_arm/stateExt:o to /yarpmotorgui0//icub/left_arm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[DEBUG]Setting a valid finder 
[... 61 lines cut ...]
[DEBUG]Initializing interfaces...
[DEBUG]Opening interfaces...
[INFO]torso iencs->getEncoders() ok!
[ERROR]Error while getting current range, part  torso  joint  0 
[ERROR]Error while getting current range, part  torso  joint  1 
[ERROR]Error while getting current range, part  torso  joint  2 
yarp: Removing output from /yarpmotorgui0//icub/head/rpc:o to /icub/head/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/head/command:o to /icub/head/command:i
yarp: Removing input from /icub/head/stateExt:o to /yarpmotorgui0//icub/head/stateExt:i
yarp: Removing output from /yarpmotorgui0//icub/right_arm/rpc:o to /icub/right_arm/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/right_arm/command:o to /icub/right_arm/command:i
yarp: Removing input from /icub/right_arm/stateExt:o to /yarpmotorgui0//icub/right_arm/stateExt:i
yarp: Removing output from /yarpmotorgui0//icub/torso/rpc:o to /icub/torso/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/torso/command:o to /icub/torso/command:i
yarp: Removing input from /icub/torso/stateExt:o to /yarpmotorgui0//icub/torso/stateExt:i
yarp: Removing output from /yarpmotorgui0//icub/left_arm/rpc:o to /icub/left_arm/rpc:i
yarp: Removing output from /yarpmotorgui0//icub/left_arm/command:o to /icub/left_arm/command:i
yarp: Removing input from /icub/left_arm/stateExt:o to /yarpmotorgui0//icub/left_arm/stateExt:i
```

---

@randaz81 commented on [Thu May 04 2017](https://github.com/robotology/yarp/issues/1196#issuecomment-299302227)

yep, it's under development. It will be fixed asap. Anyway, this issue does not generate any problems except for the message you saw when you start the gui.

aerydna, 2017-11-02:
which is the current status of this issue? @randaz81 @drdanz

traversaro, 2017-11-02:
The issue is still present, it can be easily reproduced in simulation.

randaz81, 2017-11-16:
I can take care of it in December 👍

closed 2021-03-27 by pattacini