robotology/icub-main #449: Porting Joypad to SDL2

drdanz, 2017-05-24:
We should port the joypad driver to SDL2 since SDL1 is not supported on vcpkg, I made a few tests and the only calls to replace seem to be  `SDL_JoystickName` -> `SDL_JoystickNameForIndex`.
Unfortunately there is also some weird linking issue to fix.

randaz81, 2017-11-06:
Absolutely no!

drdanz, 2017-11-07:
@randaz81 deprecate != remove.
If we don't deprecate it, new modules will be created using the old driver

randaz81, 2017-11-07:
We are not speaking of a driver. It's a user module. So far, no modules was ever removed by a robotology repo. Simply you don't use it....

drdanz, 2017-11-09:
Sorry, I thought we were speaking about a device, not a user module...

aerydna, 2017-11-09:
the problem is that old module ( replaceable by a new device ) should not be maintain, and so in this case ported to SDL2. In my opinion this issue should be closed with a `won't fix`.

commit e0f4089b (2017-11-17): SDL2 fix #449
closed 2018-02-12 by mbrunettini