robotology/icub-main #463: Polymorphism on iCub::ctrl::Filter

diegoferigo, 2017-08-09:
I'm currently developing a simulink block for wrapping `iCub::ctrl::Filter`. I'm wondering if it can be worth or has already been discussed in the past to introduce an interface e.g.

```cpp
class iCub::ctrl::IFilter {
public:
    virtual ~IFilter() = 0;
    virtual init(const yarp::sig::Vector&) = 0;
    virtual const yarp::sig::Vector& filt(const yarp::sig::Vector&) = 0;
    virtual const yarp::sig::Vector& output() const = 0;
}
```
@pattacini

pattacini, 2017-08-09:
Hi @diegoferigo 
If you reckon an I/F would help, please go on with it; I don't see any particular problem.

diegoferigo, 2017-08-11:
@pattacini 
Ok thanks, done in PR https://github.com/robotology/icub-main/pull/464

closed 2017-08-11 by pattacini