robotology/icub-main #465: Documentation of canBusSkin and embObjSkin devices

traversaro, 2017-08-16:
For people working on the skin, it would be really convenient to have a documentation of the parameters supported by the `canBusSkin` (class `CanBusSkin`) and `embObjSkin` (class `EmbObjSkin`) similar to what we have for the [`canbusanalogsensor`](http://wiki.icub.org/brain/group__canbusanalogsensor.html) and for the [`canbusinertialmtb`](http://wiki.icub.org/brain/group__canbusinertialmtb.html) devices.

traversaro, 2017-08-16:
cc @joankangro

valegagge, 2017-11-02:
Currently you can find how to configure embObjSkin in the robots configuration template files.
See
[https://github.com/robotology/robots-configuration/tree/master/iCubTemplateV3_0/hardware/skin](https://github.com/robotology/robots-configuration/tree/master/iCubTemplateV3_0/hardware/skin)

In the future, we will integrate in doxigen documentation of icub devices.

randaz81, 2017-11-06:
Similar to #477

pattacini, 2023-07-30:
The skin-related device drivers will undergo a general overhaul, which will entail that we provide documentation as well.

closed 2023-07-30 by pattacini