robotology/icub-main #466: icub::ctrl::MedianFilter::setOrder() re-initialize internal history

claudiofantacci, 2017-08-21:
I was using the `MedianFilter` and I found out that the `setOrder()` method initialize the internal history of the filter at line 326.
https://github.com/robotology/icub-main/blob/67f2f938cd57e9709932d24d9f968bb07262ad7a/src/libraries/ctrlLib/src/filters.cpp#L323-L327

A median filter can resize the filtering window without resetting the internal history. However, it may not be the case here. Is this the intended behaviour?

cc @pattacini @giuliavezzani

pattacini, 2017-08-23:
Originally, this class was not meant to deal with the dynamic tuning of the window size.

At any rate, the internal history needs to be manipulated somehow, at least in the case of an increasing order. For example, if you ask to pass from order `5` to order `11`, the history has no memory at all of the required further 6 elements that precede in time those 5 currently stored in the internal stack.

Therefore, we should manage this case to ensure a smooth transition. The story gets a bit more complicated here and possible solutions are not straightforward compared to the present code.

claudiofantacci, 2017-08-23:
@pattacini I see, it is now clear to me the viewpoint used to implement this class.

I implemented an `HistoryBuffer` class which fits to my need, you can find the implementation code [here](https://github.com/robotology/visual-tracking-control/blob/abdbbc80a0fa79ad06949810b57f49fd51856424/src/hand-tracking/src/VisualSIRParticleFilter.cpp#L459-L560) and the class definition [here](https://github.com/robotology/visual-tracking-control/blob/abdbbc80a0fa79ad06949810b57f49fd51856424/src/hand-tracking/include/VisualSIRParticleFilter.h#L33-L76).

We can discuss whether something could be added in `icub-main` as well or not in person.

closed 2017-08-23 by claudiofantacci