robotology/icub-main #467: iCubGui crashes when reading a contact from the lower leg skin

traversaro, 2017-08-22:
cc @fjandrad

traversaro, 2017-08-22:
I suspect it is related to some missing `SKIN` parameter in https://github.com/robotology/icub-main/blob/master/app/iCubGui/skeleton.ini#L17 .

fjandrad, 2017-08-22:
Isn't that parameter the  SKIN right_leg 3 in line 182?

traversaro, 2017-08-23:
The problem was actually two fold. The crash will be fixed by https://github.com/robotology/icub-main/pull/468 , but the malformed element was due to a typo in a file name contained in the `taxelPositionFiles` parameter (see https://github.com/robotology/icub-main/blob/master/app/skinGui/conf/skinManAll.ini#L31 for an example of such file). The typo went unnoticed due to the particular structure of the skinManager (see https://github.com/robotology/icub-main/blob/master/src/modules/skinManager/src/compensator.cpp#L711), that does not return an error in a case the specified file is not existing. While this defect should be fixed as well, I think this issue can be considered solved when we merge https://github.com/robotology/icub-main/pull/468 .

closed 2017-08-30 by traversaro