robotology/icub-main #475: The method embObjMotionControl::getTargetPositionsRaw(double *refs) is too slow

S-Dafarra, 2017-10-12:
I was needing to read the target positions from boards in order to debug a control module. I was using the method ``getRefPositions`` which appeared to be very slow, avoiding the program to run real-time. 

The most probable cause is that the method [``getRefPositionRaw``](https://github.com/robotology/icub-main/blob/99f94121c69a8e27fcbc090a830b264b66fc9232/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4740) is called for every joint at each time step. 

I have tried also with ``getTargetPositions`` and ``getRefVelocities`` but they have the same slow behavior. 

On the other hand ``getRefTorques`` is working fast, as expected.

Cc.
@valegagge @marcoaccame @francesco-romano @traversaro @DanielePucci

traversaro, 2017-10-12:
In general the problem is that `getRef***` is reading the reference values by using an RPC call, while all other measured values are published on the `.../stateExt:o` port. However, this applies as well as `getRefTorques`, so perhaps it is not a big problem for your application. 

Checking the code, it seems that the point is that `getRefPositionRaw` is always asking for the position to the firmware, for testing this hypothesis you can try to recompile icub-main on the robot head by commenting out this line: https://github.com/robotology/icub-main/blob/99f94121c69a8e27fcbc090a830b264b66fc9232/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L38 .

S-Dafarra, 2017-10-12:
> In general, if you want to decouple your application from the logging for debugging purpose, you can consider to use the `controlBoardDumper module.

Indeed I was using the simulink block which call [this function](https://github.com/robotology/yarp-wholebodyinterface/blob/master/src/yarpWholeBodyActuators.cpp#L1194). It was pretty convenient to use simulink since I could have done some computations on them while displaying many stuff at once.

Maybe instead of changing that line, it is safer to modify this line https://github.com/robotology/icub-main/blob/99f94121c69a8e27fcbc090a830b264b66fc9232/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4743
to `#if 0`.

valegagge, 2017-10-12:
Hi, 
I would give you some information on how data are required to firmware.

Firmware  sends each N millisec a set of data,  among wich joints' position.
[Note N is a parameter configurable in xml file, but attention to modify its value because the bandwidth we have is limited!]

While other data, among which the targetPosition, are asked to firmware directly, so a call to getTargetPosition turns into in sending a packet to the board with the request,  and, once firmware have answered, giving back the data to the application.

Some time ago we decided for some reasons , which makes no sense to examine in depth here, to  get all information from the firmware and not cache any data in embojmotioncontrol.

The long delay, you notice in get targets, I don't think is due to the time used by firmware to respond , but to the increased of the numer of packets that yarprobotinterface needs to manage both in transmission than in reception.

@S-Dafarra  So if you need to get target position only for test or debug purpose, i reccomend you to set ASK_REFERENCE_TO_FIRMWARE to 0, but I can not make sure it works becouse the code in #else branch doesn't run from a lot of time.

Moreover I would like to point out that getRefPositionRaw belong to IPositionDirect interface while getTargetPosition to IPositionControl2 interface.

At the end, the reason why the  getRefTorques is faster is that they are ref of pid and they belong to the set of periodic data sent by firmrware.

S-Dafarra, 2017-10-12:
> The long delay, you notice in get targets, I don't think is due to the time used by firmware to respond , but to the increased of the numer of packets that yarprobotinterface needs to manage both in transmission than in reception.

Can we check this somehow? 

In your opinion, could we modify the [``getTargetPositionsRaw``](https://github.com/robotology/icub-main/blob/9e5a4c026f4bcaa87819cea936bddb67d291c2e4/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4670-L4678) function so that, instead of doing a ``for`` over all the joints, query the boards only once?

barbalberto, 2017-10-13:
I think the point here is that RPC calls are intended to be used with "human" timing, not "control loop" timinng. This is because calling a RPC means that a udp/tcp message has to be created, sent over the network, then the robotInterface has to evaluate the request, create a udp/tcp message and send it back to the client.

Even if robotInterface has values cached, the network delay will still be long and undetermined. 

The `getTarget` functions are ment to debug purpose, to verify if the targets sent were received and handled correctly, but once in a while, not once in a loop.

I'd rather reconsider the requirement of calling `getTarget` so often

randaz81, 2017-10-13:
setting ASK_REFERENCE_TO_FIRMWARE to 0 does not make any sense to me, since it will work in position direct mode only. When you are using minimum jerk trajectory generator, only the firmware knows the internal reference of the PID.

francesco-romano, 2017-10-13:
@barbalberto if those functions are meant to be used for debug purpose then they should be explicitly documented.
If an interface implements methods to get references from the control board I assume that it is licit to call those methods.

The problem originated from the fact that 1 call to `getRefPositions` last `0.25 seconds` (value obtained averaging 100 calls).
@randaz81 we noticed this bug while using `getRefPositions`, but as you said this is even more important when a reference generator is acting, as the real desired joint position is unknown to the user. (Anyway I don't know if it is possible to obtain positions references if the controller is set in position mode. @S-Dafarra was pointing out all the functions that are too much slow).

@valegagge talking with @marcoaccame (of other problems) he said that the eth network is far to be congested, so I think that we may try to add those information in streaming from the boards.

valegagge, 2017-10-13:
I would like ti point out again that the getTargetPosition is the taget given by positionMove, while getRefPosition is the output of minimum jerk trajectory generator and only firmware known its value.

@S-Dafarra So if you need to cache the tagets you could use ASK_REFERENCE_TO_FIRMWARE=0, but please don't commit any change about this. This is should be done only for your debug porpouse; moreover, as @barbalberto already said, this does not allow you to reduce the delay considerably.

> In your opinion, could we modify the getTargetPositionsRaw function so that, instead of doing a for over all the joints, query the boards only once?

In my opinion it is not a good idea and it is too long explain here, so i can answer to your question face to face.

marcoaccame, 2017-10-13:
About:
>>The long delay, you notice in get targets, I don't think is due to the time used by firmware to respond , but to the increased of the numer of packets that yarprobotinterface needs to manage both in transmission than in reception.

>Can we check this somehow?

>In your opinion, could we modify the getTargetPositionsRaw function so that, instead of doing a for over all the joints, query the boards only once?

The time that embObjMotionControl takes to reply to a single blocking request sent to an ETH board is upperbound by:
` PC104::PC104TXrate + ETH_BOARD::ETH_BOARD_SETTINGS::RUNNINGMODE::period / 1000 +  PC104::PC104RXrate`
which for our robots is equal to 6 ms.

If you send more than one blocking requests,  let's say M, to a single board in a burst, the above time is not summed to 6*N because the M requests are managed in parallel and filled in the same UDP packet (typically). Hence it is still 6 ms.

In the general case, if you send more than one blocking requests,  let's say N, to a many boards (B in total) in a burst, there will be B different UDP packets sent every  `PC104::PC104TXrate = 1` ms to the B boards practically all together. The boards will reply at most with `ETH_BOARD::ETH_BOARD_SETTINGS::RUNNINGMODE::period / 1000 = 1` ms, and the PC104 will decode every  `PC104::PC104RXrate = 5` ms all the UDP packets.

Hence, the estimated upper bound of one/many blocking requests is still 6 ms.
One can reduce this upper bound to 3 ms by reducing `PC104::PC104RXrate` for instance to 1 ms.

Full details of how tuning internal communication in the robots can be found in:
https://github.com/robotology/robots-configuration/blob/master/iCubTemplateV3_0/hardware/electronics/communication-howto.txt

**To conclude**

>Can we check this somehow?

We already know how long it takes to retrieve data from an ETH board. The upper bound is:
` PC104::PC104TXrate + ETH_BOARD::ETH_BOARD_SETTINGS::RUNNINGMODE::period / 1000 +  PC104::PC104RXrate`

>In your opinion, could we modify the getTargetPositionsRaw function so that, instead of doing a for over all the joints, query the boards only once?

Actually, the queries to a single board are already transported inside the same UDP packet.

barbalberto, 2017-10-13:
When I said 'for debug', I mean that's how they are commonly used.

> If an interface implements methods to get references from the control board I assume that it is licit to call those methods.

Yes, it is legit, I'm not saying you cannot use them, only that they are slow _**by design**_, at least for RPC YARP part. Imho it is not a bug.

randaz81, 2017-10-13:
>S-Dafarra So if you need to cache the tagets you could use ASK_REFERENCE_TO_FIRMWARE=0, but please don't commit any change about this. This is should be done only for your debug porpouse; moreover, as @barbalberto already said, this does not allow you to reduce the delay considerably.

Maybe my reply was not clear, let me explain it again. Setting  ASK_REFERENCE_TO_FIRMWARE=0 is wrong, and it introduces a critical bug in the semantics the method, as explained below. Even for debug purposes, it's a bad idea to break a method for your own goal. 

> I would like ti point out again that the getTargetPosition is the taget given by positionMove, while getRefPosition is the target given with in directPosition.

This is not 100% exact and I'll like to explain it better . getRefPosition  return the position PID reference, independently from the control mode. Hence the returned value has a different meaning depending on the control mode. 
1) In directPosition, it is the target given by the user (setPosition)
2) In position mode, it is the current minimum jerk step that the pid is pursuing.

It should be clear now that setting ASK_REFERENCE_TO_FIRMWARE=0 critically breaks the system, because 2) does not work anymore. Please remove from the repo all the code related    ASK_REFERENCE_TO_FIRMWARE=0.

traversaro, 2017-10-13:
> This is not 100% exact and I'll like to explain it better . getRefPosition return the position PID reference, independently from the control mode. Hence the returned value has a different meaning depending on the control mode.

I am a bit surprised by this, because the documentation states exactly the contrary: 

> This is the dual of setPositionsRaw and shall return only values sent using IPositionDirect interface. 

See https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/libYARP_dev/include/yarp/dev/IPositionDirect.h#L69 

As most of the implementation seem to agree with the behaviour described by @randaz81 , I think it is the documentation that needs to be fixed.

randaz81, 2017-10-13:
😭 😭 😭

randaz81, 2017-10-13:
I just remembered this: https://github.com/robotology/yarp/blob/9a395f6e41e01d8d350b6a4e5e2b2f6c6f913b74/src/libYARP_dev/include/yarp/dev/IPidControl.h#L159
I'm not sure about its implementation and response time... @S-Dafarra If you want to give it a try...

francesco-romano, 2017-10-13:
@randaz81 I agree with your comments. I think that @S-Dafarra can try the `IPidControl` interface (maybe not raw?) 

Anyway I think that the results we reported are highlighting some problem with the communication as we have 250ms delay in obtaining the references from the firmware.

randaz81, 2017-10-13:
> I am a bit surprised by this, because the documentation states exactly the contrary:

Thinking about it, the behavior described in the documentation is the most clean and reasonable. Methods like **getPidReference(const PidControlTypeEnum& pidtype ...)** were introduced just to fix this ambiguity. 

Unfortunately, there are still a small bunch a modules that are using getRefPosition in the old way (to get the minimum jerk step). So, I'm going to change them to adopt the getPidReference() method.

However, some considerations mentioned above are still valid for me: reference should be always asked to fw and not cached. For example, if a user sets a position out of the joints limits, it's a fw task to clamp it to a valid value (and return the clamped value to the get() method).

I'm also thinking to return a **std::nan** value if getRefPosition() is called when control mode is not positionDirect (and extend this policy to other similar cases). This should help users to understand that something is going wrong if they do not check the boolean return value of the method (unfortunately 90% of the code does not perform this check 😞).

>As most of the actual implementations seems to agree with the behaviour described by @randaz81 , I think it is the documentation that needs to be fixed.
@valegagge and @ale-git spotted an issue about this. They are going to fix it to make the behavior complaint with documentation. 👍 👍 
it to a valid value (and return the clamped value to the get() method).

S-Dafarra, 2017-10-13:
Indeed the ``IPidControl`` interface works nicely. Thanks! I was wondering what I should use instead of ``getRefDutyCycles`` though.

In any case I still have some doubts:
- what's the difference between the two interfaces and why one is consistently faster than the other?
- given the [explanation](https://github.com/robotology/icub-main/issues/475#issuecomment-336395156) provided by @marcoaccame (btw thanks, very helpful!), I cannot figure out why those methods return every 250ms when the communication is taking at most 6ms.

valegagge, 2017-10-16:
@S-Dafarra, could you quantify 

> IPidControl interface works nicely 

please? Which is its delay?

S-Dafarra, 2017-10-16:
For the time being I can only state that it takes less than 10ms, because that is the minimum period at which I am calling that function. Basically 10ms is the smaller time slot I can measure at the moment. I don't have a direct measure of the time needed by those functions, but if necessary I could put some print here and there to have a more specific answer.

S-Dafarra, 2017-10-16:
I was eventually able to put a print only on the method. I compared ``getRefDutyCycles`` against ``getPidReferences`` (because ``getRefDutyCycles`` was the only one I was not able to get from the ``iPidControl`` interface) with 200 calls. Here the results:

| Method               | Avg [ms] | Min [ms] | Max [ms] |
|----------------------|:--------:|:--------:|:--------:|
| ``getRefDutyCycles`` |   38.7   |   20.0   |   301.78   |
| ``getPidReferences`` |    0.9   |    0.2   |    7.2   |

Apparently the 250ms were not accurate, but still 38 ms average only for that method is too high for using it repeatedly.

valegagge, 2017-10-17:
@S-Dafarra thank you!

> ...still 38 ms average only for that method is too high for using it repeatedly.

In fact, (as has been said earlier) getRef* methos should no be invoked repeatedly.

DanielePucci, 2017-10-23:
@valegagge 

> In fact, (as has been said earlier) getRef* methos should no be invoked repeatedly.

Why? References may be used in control loops for various reasons

@barbalberto 
> Yes, it is legit, I'm not saying you cannot use them, only that they are slow by design, at least for RPC YARP part. Imho it is not a bug.

This delay may cause instabilities of high level controllers. In this sense, it is a bug.

randaz81, 2017-10-24:
@barbalberto 
> Yes, it is legit, I'm not saying you cannot use them, only that they are slow by design, at least for RPC YARP part. Imho it is not a bug.

Here the main issue is due to the FW. However, your reply suggested me another topic of discussion. It would be nice to highlight, at API level, which methods are RPC and which one are streaming. Just putting a comment in a documentation seems not enough to me, but putting a RPC_* prefix to the method names is ugly too. I'm short on ideas...

S-Dafarra, 2017-10-24:
> the fastest loop is obtained asking the ref directly to the module (via yarp port)

That's true, but I think it is better what you wrote [before](https://github.com/robotology/icub-main/issues/475#issuecomment-336424885) :wink: 
> reference should be always asked to fw and not cached. For example, if a user sets a position out of the joints limits, it's a fw task to clamp it to a valid value (and return the clamped value to the get() method).
In my opinion asking the references from the module is like reading the cached value. In addition I think that the possibility of reading references directly from the robot simplifies a lot the development of modules. Suppose to have two modules that control two different sets of joints of the robot: how can you get then the references for the full robot?

Regarding 
> t would be nice to highlight, at API level, which methods are RPC

I think it is quite critic! Especially if these methods take forever to return.

>since bandwidth is limited we need to accept a tradeoff

Still it is not clear to me what the actual bottleneck is :sweat_smile:. I was not expecting the Ethernet bandwidth to be so narrow. If the problem is the serialization then, if the [above comment](https://github.com/robotology/icub-main/issues/475#issuecomment-336395156) by @marcoaccame holds, I don't understand why that time needs to be multiplied by 4, as explained [here](https://github.com/robotology/icub-main/issues/476#issuecomment-337860117).
In this context

>I already spoke with @marcoaccame, what we agreed is to speed up getRefs (multiple joints) methods by sending all the requests at once, instead of serializing them. This could bring up to a 4x speed improvements on multiple joint APIs.

it would be great :smiley:.

Finally I would like to understand why ``getPidReferences`` is fast. Does it use cached values?

randaz81, 2017-10-24:
> Finally I would like to understand why getPidReferences is fast. Does it use cached values

Pid values are broadcasted spontaneously by the FW board every cycle. 
We cannot broadcast the complete internal status of the board, because the bandwidth is limited. So we decided to broadcast them and not the targetPosition, which are thus given on demand. Hence the delay. 

>I was not expecting the Ethernet bandwidth to be so narrow. 

Very roughly, just to give an idea: 100Mbit/s = 12KBytes/ms = 240bytes per joint/ms. This is not 100% correct. There is also the skin etc. Just to give you the idea that 100Mbits are not much!

>If the problem is the serialization then, if the above comment by @marcoaccame holds, I don't understand why that time needs to be multiplied by 4.

So far multiple joint requests are serialized. I mean: you ask targetPosition for joint 0, you wait the answer, then you ask targetPosition for joint 1 etc. But Marco explained me that Eth protocol can handle multiple requests at once. For an ETH board which handles four joints, this is a 4X improvement. There are no drawbacks, just requires some modification to embObjMotionControl methods.

DanielePucci, 2017-10-24:
@randaz81 
> I think that a control loop which asks the ref to the FW is not a clean design too. All these methods are not designed to be put in a control loop: since the references are generated by someone, the fastest loop is obtained asking the ref directly to the module (via yarp port).

Given the time delay - guys, we are talking about 250 ms - the instability can be caused even if the the control loop does not ask the reference at each iteration.  @S-Dafarra [made the point](https://github.com/robotology/icub-main/issues/475#issuecomment-338936348). Also,
precluding the use of these functions - even sporadic - in a control loop casts a sinister light on their existence. In my opinion, the dogma "these functions shall be used for debug only" is ill posed since "debug" is not clear what it means. If "debug" - whatever it is - is inserted in a control loop, we risk instabilities :sweat_smile:

Anyways, thank you very much guys for the effort in reducing the execution time 👍  it starts being a fundamental point for us and it is important that the firmware developers are fully aware of the needs and bottlenecks for high level control

pattacini, 2023-07-30:
Long discussion. I think people managed to survive and get around this time limitation.

closed 2023-07-30 by pattacini