robotology/icub-main #476: a few methods of motor control interfaces are very slow

massimoregoli, 2017-10-17:
I experienced this issue on the purple iCub, while the same code on the black iCub is working fine. Also, it is exactly the same code that was working about 1-2 months ago on the purple iCub.

Basically my code relies on a few motor control interfaces, and I just noticed that now the following calls are very very slow (not even seconds but minutes...):
- `IVelocityControl::stop()`
- `IPositionControl::checkMotionDone(...)`
- `IControlMode2::setControlMode(...)`

Restarting motors/pc104 doesn't help. Do you have any idea why this could happen?

pattacini, 2017-10-17:
cc @marcoaccame @valegagge @randaz81

randaz81, 2017-10-17:
minutes for IControlMode2::setControlMode() ???

massimoregoli, 2017-10-17:
yes, but is definitely unusual also for the other calls right?

traversaro, 2017-10-17:
On which joints are you experiencing this? 
When the methods eventually return, they return successfully (i.e. the control mode was actually changed)? 
Could you attach the full log of the `yarprobotinterface` when you call the method (and when eventually returns)?

massimoregoli, 2017-10-17:
`stop()` and `checkMotionDone()` are called on the whole arm (without specifying any joint), while `setControlMode()` is called on the joints 9 and 13 of the arm (proximal joints of the thumb and the middle finger).
The methods eventually return successfully. I'll attach the log as soon as I possible.

massimoregoli, 2017-10-17:
Ok, here is the log of `yarprobotinterface `about 2 minutes after that `IVelocityControl::stop()` is colled (no warnings/errors are shown before):

```
6818    1628,360558    ERROR    embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD right_arm-eb4-j4_15 IP 10.0.1.4 for nv ID32 = 0x01000b0b -> IND = 11, TAG = eoprot_tag_mc_joint_status_target
6819    1628,360644    ERROR    embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD right_arm-eb4-j4_15 IP 10.0.1.4 joint  11
6820    1628,390850    ERROR    Received an answer message nobody is waiting for (eoThreadFifo::pop)
6821    1628,390907    ERROR    Received an answer message nobody is waiting for (MCmutex_post)
6822    1628,390924    WARNING    while releasing mutex for IP 10.0.1.4 and NV ID32 = 0x01000b0b -> IND = 11, TAG = eoprot_tag_mc_joint_status_target
```

Anyway at this point the method has not returned yet. After 30 seconds, as soon as it returns, I get:
```

6887    1660,641962    WARNING     from BOARD 10.0.1.4 (right_arm-eb4-j4_15), src LOCAL, adr 0, time 1276s 582m 919u: (code 0x0000000d, par16 0x0215 par64 0x000c001901720007) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
6888    1660,652126    WARNING     from BOARD 10.0.1.4 (right_arm-eb4-j4_15), src LOCAL, adr 0, time 1276s 583m 676u: (code 0x0000000b, par16 0x01e3 par64 0x0019017200070215) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
```

I also run the method `IPositionControl::checkMotionDone(...)`, it takes about 10 minutes to return! (I guess it calls many times `checkMotionDone(joint j,...)` on each joint. After it returns, I get this:

```
8116    2267,276589    ERROR    embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD right_arm-eb3-j0_3 IP 10.0.1.3 for nv ID32 = 0x0100010b -> IND = 1, TAG = eoprot_tag_mc_joint_status_target
8117    2267,276634    ERROR    embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD right_arm-eb3-j0_3 IP 10.0.1.3 joint  1
8119    2267,578668    ERROR    embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD right_arm-eb3-j0_3 IP 10.0.1.3 for nv ID32 = 0x0100000b -> IND = 0, TAG = eoprot_tag_mc_joint_status_target
8120    2267,578705    ERROR    embObjMotionControl::getTargetPositionRaw() could not read reference pos for  BOARD right_arm-eb3-j0_3 IP 10.0.1.3 joint  0
8121    2267,578715    ERROR    Received an answer message nobody is waiting for (eoThreadFifo::pop)
8122    2267,578724    ERROR    Received an answer message nobody is waiting for (MCmutex_post)
8123    2267,578731    WARNING    while releasing mutex for IP 10.0.1.3 and NV ID32 = 0x0100010b -> IND = 1, TAG = eoprot_tag_mc_joint_status_target
8124    2267,578738    ERROR    Received an answer message nobody is waiting for (eoThreadFifo::pop)
8125    2267,578745    ERROR    Received an answer message nobody is waiting for (MCmutex_post)
8126    2267,578752    WARNING    while releasing mutex for IP 10.0.1.3 and NV ID32 = 0x0100000b -> IND = 0, TAG = eoprot_tag_mc_joint_status_target
```

marcoaccame, 2017-10-17:
there is something very much unusual.
tomorrow morning i will come over to have a look and collect more evidence.
@massimoregoli, are you available to show me the conditions of failure?

massimoregoli, 2017-10-18:
Sure, let me know when

marcoaccame, 2017-10-18:
It seems that the method `embObjMotionControl::getTargetPositionsRaw()` was called every 15 ms for the lower arm which has 12 joints. 

It is a problem similar to #475.

It should be enough to reduce the calling rate.

francesco-romano, 2017-10-18:
@marcoaccame what are the function calls of @massimoregoli code? Because if they are only "stop", "checkMotionDone", etc, those calls are pretty standard and have always worked.
If that is the case, I am more prone to think that something has changed at the firmware level

cc. @pattacini @traversaro

massimoregoli, 2017-10-18:
@francesco-romano I noticed the delay on the calls I listed above, that is:

- `IVelocityControl::stop()`
- `IPositionControl::checkMotionDone(...)`
- `IControlMode2::setControlMode(...)`

Anyway, when these method are called, at the same time a separate thread collects some data from the robot. Specifically, it calls `IPositionControl2::getTargetPositions(...)` every 15 ms, and this probably caused the problem. When I wrote the code such data was cached on the pc104 by `yarprobotinterface`, so the call was very fast. Now, as I was told, it is not cached anymore and each time the data has to be retrieved directly from the arm. So I am definitely overusing such method, which generates too much data traffic and affects the above mentioned calls. I tried to temporarily comment it out and actually `IVelocityControl::stop()` now works fine.
I have now to change some code to be able to run a full test (I have to remove the dependency on `IPositionControl2::getTargetPositions(...)`), then if everything works I'll close the issue.

Thanks to @valegagge and @marcoaccame for their help!

massimoregoli, 2017-10-18:
Ok everything works now!

closed 2017-10-18 by massimoregoli

marcoaccame, 2017-10-19:
Hi @francesco-romano, I have seen that @massimoregoli has already answered. 

I am happy that now things work. Usual standard calls keep on working if they are used correctly.

To clear you fears, let me add that nothing relevant has changed at firmware level. 

What was instead changed (about < 1 year) is the way some values such as references, setpoints, etc are managed at robotinterface level.

These values are generated by an external application and sent down to boards via robotinterface. If another application wants to know them, it performs a call the likes of IPositionControl2::getTargetPositions().
  
Some time ago, these values were cached into the PC104, hence the call IPositionControl2::getTargetPositions() was extremely quick because it just had to read a value in RAM. 

However, the caching mechanism was not correct for several reason. For instance, what if the target position was out of the allowed limits? What if the controller inside the board caused saturation, underflow or quantization errors?

Hence, @randaz81 and @valegagge designed another mechanism. Now functions such as IPositionControl2::getTargetPositions() always ask in blocking mode directly to the relevant board. Because only the board has the full knowledge of if and how the "order from an external application should be executed".  

This mechanism much improves reliability but, I understand, has had an impact on response time: from fractions of milli to multiples of milliseconds. 

As an example see #475, where @S-Dafarra has measured the time of getRefDutyCycles(). In this case, there are njoints = ? consecutive blocking requests sent directly to the board for a data structure tagged eoprot_tag_mc_motor_status_basic which is 20 bytes long. It takes 38.7 ms in average.

In case of IPositionControl2::getTargetPositions() on the lower arm there are njoints = 16 blocking request of a data structure tagged eoprot_tag_mc_joint_status_target which is 24 bytes long. Hence time is comparable with 40 ms. 

If some time ago the IPositionControl2::getTargetPositions() could be called every 15 ms, now that cannot be done anymore.

Having said that, if anybody has a strong equirement to call methods such as IPositionControl2::getTargetPositions() at a faster pace, I think that we can find a way to guarantee both reliability and speed. 
For instance, we can still use a cached value to get() BUT we must validate the value by asking to the board every time an external application changes it with a set().

If there is this requirement, please let us discuss together in another issue or dedicated meeting with at least @randaz81, @valegagge, @francesco-romano, @traversaro, and me (others?).

francesco-romano, 2017-10-19:
Thanks @marcoaccame.

Let's collect the requirements and if needed we can meet to discuss this f2f

randaz81, 2017-10-19:
> eoprot_tag_mc_joint_status_target which is 24 bytes long. Hence time is comparable with 40 ms.

So required time is proportional to message length? Are all these 24 bytes really necessary to get target information? Can we get performance improvements by using a smaller data structure? On CAN robot the answer message was  just 8 bytes long.

marcoaccame, 2017-10-19:
Hi @randaz81, time of a direct blocking request to an ETH board is not proportional to message length.

As long as the requested bytes stay inside a UDP packet, time depends on: frequency of the TX and RX threads in PC104 (xml configurable: now at 1ms and 5 ms), on their regularity of execution (Linux RT and high priority of threads), on control loop speed of the ETH boards (now at 1ms).

Time of asking the info for a single joint is *at most* 3-7 ms (as I explained in #475) irrespectively of the size. Yes, we may ask smaller chunks of data, but there would be no improvement. 

We may speed things up in other ways at robotinterface level. 

For instance: 
1. by sending parallel requests to the ETH boards and not using serialization as it is done now in methds which return values for multiple joints. Serialization multiplies the required ms typically by 4 (by 12 in the lower arm of the purple).
2. using a smart caching as I wrote in the previous post.
3. something at higher layer than the device? what's your opinion @barbalberto ?