robotology/icub-main #477: Documentation of available calibrators

randaz81, 2017-11-06:
We currently have a bunch of calibrators, identified by an int value, whose parameters are undocumented. e.g. 3 for absolute AEA encoders etc.
We should produce a doc, describing the accepted parameters for each calibrator type and the robots which use them.

valegagge, 2017-12-04:
Hi,
in attachment you can find the first draft of document about calibrations available on iCub and R1 robots.

Currently it contains only some of used calibrations, but asap will complete the document.
[iCub_R1_calibrations.pdf](https://github.com/robotology/icub-main/files/1527411/iCub_R1_calibrations.pdf)

plinioMoreno, 2017-12-06:
Hi @randaz81 and @valegagge,

I guess Vizzy is also using the calibration, but the old types of calibration. Is there a document like the one in this issue for [Vizzy's](https://github.com/robotology/robots-configuration/tree/master/vizzy/calibrators) calibration files?

Thanks,

valegagge, 2018-01-22:
I transformed the document I had attached to a wiki page.

I inserted the chapter 3.3 "Calibration types available on iCub and R1 robots" in the iCub manual.

See [https://wiki.icub.org/wiki/Calibration_types_available_on_iCub_and_R1_robots](url)

pattacini, 2021-03-27:
See https://icub-tech-iit.github.io/documentation/icub_r1_calibration_types/icub_r1_calibration_types/#the-available-calibrations-in-details.

closed 2021-03-27 by pattacini