robotology/icub-main #478: camCalib crashes (mostly left eye)

claudiofantacci, 2017-11-28:
While working on Blacky, the (left eye) `camCalib` keeps crashing regularly after some times.
Together with @giuliavezzani, we have experienced this issue while extensively using images (i.e. a lot of connection) from the left eye.

This issue is to keep track to this issue and eventually to trigger discussion.
@Nicogene and I will do some tests trying to log data and understand where the problem is.

randaz81, 2017-11-28:
Please specific also the carrier used for connections.
I remember that I saw similar crashes sometime ago using exotic stuff such as `shmem`... 
And I'm not 100% sure about mjpeg, too.

Nicogene, 2017-11-28:
Input connections : `tcp`
Output connections : `udp`

Why this choice is a mystery 😅

pattacini, 2017-11-28:
@Nicogene `tcp` is applied to a single connection (`framegrabber` ➡️  `camCalib`), whereas `udp` is applied to multiple connections since `camCalib` serves many downstream modules.

I think this was the original rationale, perhaps ;)

Nicogene, 2017-11-28:
Ok for `tcp` (maybe it is better `fast_tcp` ? Do we need a syncro with the sender?), but the most efficient carrier for the **one-to-many** connections is `mcast` that at the time in which the xml was created was full of 🐞 but now it is a 🚀  😉

pattacini, 2017-11-28:
I agree with you @Nicogene for `mcast`.
As of now, please let's stick to `tcp`.

Nicogene, 2017-11-28:
The first thing to do is to check, when this problem happens, if `/icub/cam/left` is actually streaming images. If doesn't, it is a problem in dragonfly2 cameras (HW ? SW?), otherwise it is `camCalib` fault  🕵️

claudiofantacci, 2017-11-28:
> Ok for tcp (maybe it is better fast_tcp ? Do we need a syncro with the sender?), but the most efficient carrier for the one-to-many connections is mcast that at the time in which the xml was created was full of 🐞 but now it is a 🚀 😉

This would also be worth trying, but 

> As of now, please let's stick to tcp.

This is also valid, at least in the short term, but @Nicogene, @giuliavezzani and I can do some tests with our setting to check this out.

claudiofantacci, 2018-06-12:
Do we have any update on this?
@Nicogene

closed 2021-03-27 by pattacini