robotology/icub-main #484: Extend Python bindings

Tobias-Fischer, 2018-01-11:
This issue is a follow-up on https://github.com/robotology/icub-main/pull/479 and https://github.com/robotology/icub-main/issues/302. 

In https://github.com/robotology/icub-main/pull/479 the linker error reported in https://github.com/robotology/icub-main/issues/302 was fixed, and the bindings now wrap the `iKin` library. Using the same approach, it should be straightforward to also wrap the following other libraries (maybe during an IFI day?!?):
- `actionPrimitives`
- `ctrlLib`
- `iDyn`
- `learningMachine`
- `optimization`
- `perceptiveModels`
- `skinDynLib`

However, I am not sure about the plugins within `icubmod`, as they are yarp plugins (opposed to the libraries listed above), which has the implication that their header files are not being installed at the moment (which is the minimum-minimum requirement needed for wrapping).

/cc @pattacini @traversaro @drdanz @towardthesea

traversaro, 2018-01-11:
> However, I am not sure about the plugins within `icubmod`, as they are yarp plugins (opposed to the libraries listed above), which has the implication that their header files are not being installed at the moment (which is the minimum-minimum requirement needed for wrapping).

The headers of the YARP devices do not need to be installed in order to be used from C++ code. 
This is one of the main advantage of using classes loaded at runtime through a factory such as `yarp::dev::PolyDriver`. The same advantage applies to the binding, with the caveat that is tipically needed to explicitly expose the specific C++ interface with which you want to interface with the class. As the `ICartesianController` and `IGazeController` interface are wrapped in the bindings at the YARP level ( https://github.com/robotology/yarp/blob/master/bindings/yarp.i#L769 ) there should be no problem in using this devices from the bindings.

Tobias-Fischer, 2018-01-11:
Thanks @traversaro. Therefore, I guess only wrapping of the other libraries mentioned above is desirable. In theory, this can be solved by simply copying the (names of the) relevant header files to `bindings.i`. I'll have a look at this at an IFI day / when I have some spare time.

towardthesea, 2018-01-11:
Maybe I can give a hand on this :smile: 
However, it may be useful if somebody can suggest an easy way to test if the new fixed `binding.i` can really work.

Tobias-Fischer, 2018-01-11:
I suggest to setup tests, in a similar fashion to https://github.com/robotology/yarp/tree/master/bindings/tests/python
Within the test, you can simply try to instantiate and use some of the commonly used classes which are meant to be wrapped.
As a bonus, you could try to add a test for the `ICartesianController` and `IGazeController` interface wrapped by yarp (although it might be better to add the tests for these interfaces directly in yarp rather than here).

towardthesea, 2018-08-17:
After the merge #544 , `icub-main` still have some following libraries need to be wrapped by `icub-bindings`:
- actionPrimitives
- iDyn: I got an error compiling binding for this library, in particularly [iDynBody](https://github.com/robotology/icub-main/blob/master/src/libraries/iDyn/include/iCub/iDyn/iDynBody.h). Maybe @traversaro can help a bit
- learningMachine: this one is crazily scary (~20 source file) 
- optimization
- perceptiveModels

I will find some time to finish this one soon!
Cheers,
Phuong

pattacini, 2018-08-18:
@towardthesea it is not worth exposing bindings for `iDyn` since this library is deprecated in favor of `iDynTree`.

Similarly, I honestly doubt it might be of help working out:
- `actionPrimitives`
- `learningMachine`
- `perceptiveModels`

Differently, having access to services provided by `optimization` might be worthy.

towardthesea, 2018-08-18:
Thanks @pattacini for insight advice! So there should be another `PR` for `optimization` library and we can close this issue :smile:

traversaro, 2018-08-18:
+1 in not wrapping iDyn

closed 2021-03-27 by pattacini