robotology/icub-main #492: skinManager fails to open

melszeen, 2018-02-07:
Hello,

I am working with the iCub Simulator, and when trying to open skinManager, I get this error:

```
yarp: Port /skinManager/rpc active at tcp://10.217.131.124:10052/
yarp: Port /skinManager/monitor:o active at tcp://10.217.131.124:10053/
yarp: Port /skinManager/info:o active at tcp://10.217.131.124:10054/
7 input ports found in the configuration file
[INFO]Input port: /icub/skin/left_hand  -> Output port: /icub/skin/left_hand_comp
yarp: Port /icub/skin/left_hand_comp active at tcp://10.217.131.124:10055/
yarp: Port /skinManager0/input active at tcp://10.217.131.124:10056/
yarp: Port /skinManager0/input/rpc:o active at tcp://10.217.131.124:10057/
[ERROR]AnalogSensorClient::open() error could not connect to /icub/skin/left_hand
yarpdev: ***ERROR*** driver <analogsensorclient> was found but could not open
[INFO][/icub/skin/left_hand]: Device not available.
[INFO]Input port: /icub/skin/left_forearm  -> Output port: /icub/skin/left_forearm_comp
yarp: Port /icub/skin/left_forearm_comp active at tcp://10.217.131.124:10058/
yarp: Port /skinManager1/input active at tcp://10.217.131.124:10059/
yarp: Port /skinManager1/input/rpc:o active at tcp://10.217.131.124:10060/
[ERROR]AnalogSensorClient::open() error could not connect to /icub/skin/left_forearm
yarpdev: ***ERROR*** driver <analogsensorclient> was found but could not open
[INFO][/icub/skin/left_forearm]: Device not available.
[INFO]Input port: /icub/skin/left_arm  -> Output port: /icub/skin/left_arm_comp
yarp: Port /icub/skin/left_arm_comp active at tcp://10.217.131.124:10061/
yarp: Port /skinManager2/input active at tcp://10.217.131.124:10062/
yarp: Port /skinManager2/input/rpc:o active at tcp://10.217.131.124:10063/
[ERROR]AnalogSensorClient::open() error could not connect to /icub/skin/left_arm
yarpdev: ***ERROR*** driver <analogsensorclient> was found but could not open
[INFO][/icub/skin/left_arm]: Device not available.
[INFO]Input port: /icub/skin/right_hand  -> Output port: /icub/skin/right_hand_comp
yarp: Port /icub/skin/right_hand_comp active at tcp://10.217.131.124:10064/
yarp: Port /skinManager3/input active at tcp://10.217.131.124:10065/
yarp: Port /skinManager3/input/rpc:o active at tcp://10.217.131.124:10066/
[ERROR]AnalogSensorClient::open() error could not connect to /icub/skin/right_hand
yarpdev: ***ERROR*** driver <analogsensorclient> was found but could not open
[INFO][/icub/skin/right_hand]: Device not available.
[INFO]Input port: /icub/skin/right_forearm  -> Output port: /icub/skin/right_forearm_comp
yarp: Port /icub/skin/right_forearm_comp active at tcp://10.217.131.124:10067/
yarp: Port /skinManager4/input active at tcp://10.217.131.124:10068/
yarp: Port /skinManager4/input/rpc:o active at tcp://10.217.131.124:10069/
[ERROR]AnalogSensorClient::open() error could not connect to /icub/skin/right_forearm
yarpdev: ***ERROR*** driver <analogsensorclient> was found but could not open
[INFO][/icub/skin/right_forearm]: Device not available.
[INFO]Input port: /icub/skin/right_arm  -> Output port: /icub/skin/right_arm_comp
yarp: Port /icub/skin/right_arm_comp active at tcp://10.217.131.124:10070/
yarp: Port /skinManager5/input active at tcp://10.217.131.124:10071/
yarp: Port /skinManager5/input/rpc:o active at tcp://10.217.131.124:10072/
[ERROR]AnalogSensorClient::open() error could not connect to /icub/skin/right_arm
yarpdev: ***ERROR*** driver <analogsensorclient> was found but could not open
[INFO][/icub/skin/right_arm]: Device not available.
[INFO]Input port: /icub/skin/torso  -> Output port: /icub/skin/torso_comp
yarp: Port /icub/skin/torso_comp active at tcp://10.217.131.124:10073/
yarp: Port /skinManager6/input active at tcp://10.217.131.124:10074/
yarp: Port /skinManager6/input/rpc:o active at tcp://10.217.131.124:10075/
[ERROR]AnalogSensorClient::open() error could not connect to /icub/skin/torso
yarpdev: ***ERROR*** driver <analogsensorclient> was found but could not open
[INFO][/icub/skin/torso]: Device not available.
[ERROR][CompensationThread]  Compensator /icub/skin/left_hand did not open correctly. Removing the port.
[ERROR][CompensationThread]  Compensator /icub/skin/left_forearm did not open correctly. Removing the port.
[ERROR][CompensationThread]  Compensator /icub/skin/left_arm did not open correctly. Removing the port.
[ERROR][CompensationThread]  Compensator /icub/skin/right_hand did not open correctly. Removing the port.
[ERROR][CompensationThread]  Compensator /icub/skin/right_forearm did not open correctly. Removing the port.
[ERROR][CompensationThread]  Compensator /icub/skin/right_arm did not open correctly. Removing the port.
[ERROR][CompensationThread]  Compensator /icub/skin/torso did not open correctly. Removing the port.
[ERROR][CompensationThread] No input port left. Stopping the thread.
yarp: A thread failed to start with error code: 0
[ERROR][SkinManager] Could not start the compensator thread. 
[INFO]RFModule failed to open.
```
and when checking the port list, the input ports (those with error) were not available.

Thanks in advance!

pattacini, 2018-02-11:
Hi @melszeen 

The simulator opens up ports with the prefix `/icubSim` as the following:

```
registration name /icubSim/skin/left_arm_comp ip 10.0.75.1 port 10014 type tcp
registration name /icubSim/skin/left_forearm_comp ip 10.0.75.1 port 10016 type tcp
registration name /icubSim/skin/left_hand_comp ip 10.0.75.1 port 10003 type tcp
registration name /icubSim/skin/left_hand_comp/rpc:i ip 10.0.75.1 port 10004 type tcp
registration name /icubSim/skin/right_arm_comp ip 10.0.75.1 port 10015 type tcp
registration name /icubSim/skin/right_forearm_comp ip 10.0.75.1 port 10017 type tcp
registration name /icubSim/skin/right_hand_comp ip 10.0.75.1 port 10005 type tcp
registration name /icubSim/skin/right_hand_comp/rpc:i ip 10.0.75.1 port 10006 type tcp
registration name /icubSim/skin/torso_comp ip 10.0.75.1 port 10018 type tcp
registration name /icubSim/skinManager/skin_events:o ip 10.0.75.1 port 10013 type tcp
```

Therefore, you have to tell the `skinManager` to connect to the correct ports. This can be done by means of proper configuration files.

An example is launching:
```sh
$ skinManager --from skinManSim.ini
```

melszeen, 2018-02-13:
Okay thanks for your help!

closed 2018-02-13 by melszeen