robotology/icub-main #497: PWM measurements are streamed from the robot in percentage over the full scale, but all the other parameters in API and in configuration files are using PWM as raw integers

traversaro, 2018-02-23:
See https://github.com/robotology/yarp/issues/1556#issuecomment-367955842 .

traversaro, 2018-02-23:
@valegagge wrote:

> @traversaro you are right!
> 
> We need to update API (or use new interface) in order to use PWM in %.
> 
> @nunoguedelha
> You can use the full scale in the configuration files to convert from raw to %.
Use parameterfullscalePWM; see https://github.com/robotology/robots-configuration/blob/8423a4d9ef054ed579b67dccac1cc34ec9105e5c/iCubTemplates/iCubTemplateV4_0/hardware/mechanicals/body_part-ebX-jA_B-mec.xml#L12 .

> As a temporary solution, I can add a remote variable for fullscalePWM, but I can't implement it immediately

randaz81, 2018-05-03:
fixed by https://github.com/robotology/yarp/pull/1612

closed 2018-05-03 by randaz81