robotology/icub-main #501: iCub_SIM simulator not work in docker robotology/icub

niudongzone, 2018-03-20:
Hello bro!
I have downloaded the docker file by `docker pull robotology/icub` command,and it is work well when i run `yarpserver` but when i run `iCub_SIM`,i got the error below:
```
[INFO]Adding video texture face
yarp: Port /icubSim/texture/face active at tcp://172.17.0.2:10049/
[INFO]Adding video texture screen
yarp: Port /icubSim/texture/screen active at tcp://172.17.0.2:10050/
[INFO]Adding video texture sky
yarp: Port /icubSim/texture/sky active at tcp://172.17.0.2:10051/
[DEBUG]***** OdeSdlSimulation::simLoop 
Segmentation fault (core dumped)
```
Is it because of docker?
Waiting for response,thanks!

pattacini, 2018-03-20:
Not completely sure but @diegoferigo might be of some help in this respect.

diegoferigo, 2018-03-20:
Can you please post how you create the container? If you want to use GUIs, you need to instruct docker how to access the host's Xserver.

niudongzone, 2018-03-21:
Thanks, @diegoferigo. I guess it must be the reason you mentioned so i made some configurations according to [can-you-run-gui-apps-in-a-docker-container](https://stackoverflow.com/questions/16296753/can-you-run-gui-apps-in-a-docker-container). But it is still not work so maybe i will try setup on my ubuntu system rather than docker.

diegoferigo, 2018-03-21:
There are many ways to share Xserver. Don't do complicated ways, [here](https://github.com/diegoferigo/dockerfiles/blob/83eaff1bc4cf6814fbc67376d88932ef4c40d248/ROS/README.md) you can find an old setup of my ROS image. Now I switched to a more secure way, but the command you find in the link is the easiest way to get it working.

diegoferigo, 2018-03-26:
I had some time to look into this issue, and with the image you have I managed to get `iCubGui` successfully running. Here below the command I used:

```bash
$ xhost +local:root
$ docker run -it --rm \
      -e DISPLAY \
      -v /tmp/.X11-unix:/tmp/.X11-unix:rw \
      robotology/icub \
      iCubGui
$ xhost -local:root
```

The image works out-of-the-box for intel-based machines. If your laptop has an Nvidia card (as mine) you need a more complex configuration, which involves [`nvidia-docker`](https://github.com/NVIDIA/nvidia-docker/). [Here is a pointer](https://github.com/NVIDIA/nvidia-docker/issues/136#issuecomment-232755805) to a simple test for future reference.

closed 2019-09-22 by pattacini

arfan786, 2021-02-19:
How can create image in icub simulator  please  help me in code