robotology/icub-main #504: iCubGui have BottleImpl reader failed when visual objects

caicaixia, 2018-03-21:
I tried to visualize a red ball in the iCubGui,  I did like what wiki said. However, the ball is only shown and didn't get the position update.

I also tried direct:
`yarp rpc /iCubGui/objects`
`>>object ball 50 50 50 -300.000 -0.001  100.417 0.0 0.0 0.0 255 0 0 1.0 `
then it shows some errors:

[image]

1: Why this has happened?
2: How can I show the "coordination frame" above iCub head?

Thanks.

pattacini, 2018-03-21:
Hi @caicaixia 

To talk to `iCubGui` for what pertains displaying objects, you have to use `yarp write ... /iCubGui/objects` and not `yarp rpc /iCubGui/objects` since iCubGui won't provide you back with any reply. That's the reason why you saw the error message.

Then, to create/update the objects positions, just look at the following:
- iCubGui [main page](http://wiki.icub.org/iCub/main/dox/html/group__icub__gui.html)
- Example of [code snippet](https://github.com/robotology/d4c/blob/master/lib/src/d4c_server.cpp#L2058-L2091)

Finally, if you connect the iCubGui straight to the robot / simulator, you'll see the output of the IMU popping up above the iCub's head.

caicaixia, 2018-03-22:
Hello @pattacini 
Thanks for your reply.  

1.  Yes, if I directly use `yarp write ... /iCubGui/objects` sent commands, the object position is changed.  However,  If I put the example code like the tutorial,  the object is never changed, it stays in the same position.
My codes are same: 
```
yarp::os::BufferedPort<yarp::os::Bottle> outportGui;

if (outportGui.getOutputCount()>0)
        {
            Vector rpy=CTRL_RAD2DEG*dcm2rpy(axis2dcm(od));

            Bottle &obj=outportGui.prepare();
            obj.clear();

            obj.addString("object"); // command to add/update an object
            obj.addString("red_ball");
            // object dimensions in millimiters
            // (it will be displayed as an ellipsoid with the tag "red_ball")
            obj.addDouble(50.0);
            obj.addDouble(50.0);
            obj.addDouble(50.0);
            // object position in millimiters
            // reference frame: X=fwd, Y=left, Z=up
            obj.addDouble(1000.0*xd[0]);
            obj.addDouble(1000.0*xd[0]);
            obj.addDouble(1000.0*xd[0]);
            obj.addDouble(rpy[0]);                      // rotX [deg]
            obj.addDouble(rpy[1]);                      // rotY [deg]
            obj.addDouble(rpy[2]);
            // object color (0-255)
            obj.addInt(255);
            obj.addInt(0);
            obj.addInt(0);
            // transparency (0.0=invisible 1.0=solid)
            obj.addDouble(1.0);

//            outportGui.write(obj);
            outportGui.writeStrict();
            std::cout <<"updating gui item red ball!\n";
        }
```
[image]

2.
>  Finally, if you connect the iCubGui straight to the robot/simulator, you'll see the output of the IMU popping up above the iCub's head.

What means connect the iCubGui straight to the robot/simulator? 
I did like this,  and it does not show the IMU:
```
  <module>
      <name>iCubGui</name>
      <parameters>--xpos 1500 --ypos 50 --width 370</parameters> 
      <node>localhost</node>
      <tag>iCubGui</tag>
  </module>

  <connection>
      <from>/redBall/gui:o</from>
      <to>/iCubGui/objects</to>
      <protocol>tcp</protocol>
  </connection>

<connection>
    <from>/icubSim/head/state:o</from>
    <to>/iCubGui/head:i</to>
    <protocol>udp</protocol>
  </connection>

  <connection>
    <from>/icubSim/torso/state:o</from>
    <to>/iCubGui/torso:i</to>
    <protocol>udp</protocol>
  </connection>

  <connection>
    <from>/icubSim/left_arm/state:o</from>
    <to>/iCubGui/left_arm:i</to>
    <protocol>udp</protocol>
  </connection>

  <connection>
    <from>/icubSim/right_arm/state:o</from>
    <to>/iCubGui/right_arm:i</to>
    <protocol>udp</protocol>
  </connection>
```

pattacini, 2018-03-22:
Try to add up the following connection to your application:
```xml
<connection>
  <from>/icubSim/inertial</from>
  <to>/iCubGui/inertial:i</to>
  <protocol>udp</protocol>
</connection>
```

pattacini, 2018-03-28:
Hi @caicaixia 
Any news on this?
Otherwise, I'd close it up.

caicaixia, 2018-03-29:
hello @pattacini.  Thanks a lot. That is working.

May I ask one more question: How can I visualize the coordination frames for the different part in iCubGUI: e.g. hand?

pattacini, 2018-03-29:
`iCubGui` does not display frames attached to robot parts. To this end, you may consider exploring as alternatives `rviz` and/or the new iCub simulator based on `Gazebo`. The relevant repository is https://github.com/robotology/gazebo-yarp-plugins.

closed 2018-03-29 by pattacini