robotology/icub-main #511: Travis on macOS is failing

traversaro, 2018-04-18:
Something changed ~20 days ago, and now the Travis job on macOS is failing with the following error:
~~~
==> Pouring python@2-2.7.14_3.sierra.bottle.1.tar.gz
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink bin/2to3-2
Target /usr/local/bin/2to3-2
is a symlink belonging to python. You can unlink it:
  brew unlink python
To force the link and overwrite all conflicting files:
  brew link --overwrite python@2
To list all files that would be deleted:
  brew link --overwrite --dry-run python@2
~~~

See https://travis-ci.org/robotology/icub-main/jobs/364077632

Tobias-Fischer, 2018-04-19:
See https://github.com/Homebrew/homebrew-core/issues/26358 for a potential fix.

Tobias-Fischer, 2018-08-11:
Fixed with https://github.com/robotology/icub-main/commit/70af98fcda594207005d750f4332d91d82bc100d

closed 2018-08-11 by Tobias-Fischer