robotology/icub-main #522: iCub_SIM control stability problem

pattacini, 2018-05-27:
Following up the latest modifications that yarp's been undergoing, `iCub_SIM` does no longer seem to be running properly under Windows in terms of control stability.

Find below an example where the head moves responding to the `iKinGazeCtrl` commands. It is evident how the control stability is somewhat compromised.

[image]

Did anything change with respect to the low level PID gains / control?

Weirdly, the control seems to be ok under Linux instead.
I can provide more details.

cc @randaz81

pattacini, 2018-06-28:
This is becoming quite an important issue that needs our attention!

lornat75, 2018-07-13:
do we have latency tests we can run on windows?

lornat75, 2018-07-13:
@Nicogene

Nicogene, 2018-07-13:
We should investigate which are the differences from linux and windows.
The only changes made on the communication part are the one of `TCP_CORK`, I don't know if in windows is defined, if not defined the tcp carrier is the "old" one.

lornat75, 2018-07-13:
it would not harm to run a few latency tests win2win, win2linux and linux2win

pattacini, 2018-08-30:
I've just done a quick test again on this and "_automagically_" the problem has disappeared: everything's working well again.

Is anyone aware of any possible dev/fix that sorted this out? @lornat75 @drdanz @Nicogene @randaz81

closed 2018-08-30 by pattacini

Nicogene, 2018-08-30:
Unfortunately I'm not, it came from nowhere and returned to nowhere :sweat: :beetle: :alien: