robotology/icub-main #530: skin wrapper complains if we remove from device's xml file a not-mounted mtb

marcoaccame, 2018-06-14:
In here I document a behavior which I saw on iCubGenova02 and for which I suggest to find a solution.

In the left leg of robot iCubGenova02 some mtb boards are written in the /hardware/skin/left_leg-eb10-skin.xml file but are not mounted on the robot (or they are dead). They are  CAN2:13 and CAN2:12.

If we specify in the .xml file to check the board presence, the device complains that does not find the two boards and does not start, resulting in a failure of yarprobotinterface.

This is the correct behavior which we also have for other devices.

At this point, one:
* would ask icub-support for a fix of the robot, 
* but in the meantime would ask to remove the two addresses from the xml file

However, here is a strange behavior:
If we remove these two addresses from the xml file, then everything starts correctly but the wrapper complains that it does not receive the correct number of taxels. The result is that yarprobotinterface is unusable due to the continuous prints.

What I think is that the device should send data in such a way to avoid complains from the wrapper.

julijenv, 2018-06-14:
Hi @marcoaccame ,
unfortunately icub-support can't do anything about it because MTB board ID 12 and 13 or actually the 2 mtb boards attached to the feet, and as a matter of fact they are not mounted neither present from every place I looked for so I guess that they dont use it and consequently the general.xml should have those parts of skin unabled un til they want to reuse them.

valegagge, 2018-06-19:
I checked the source code of wrapper and I think that this is not a bug, but it due to a wrong configuration of wrapper in xml file. if one or more boards are not available, it is necessary update wrapper's configuration accordingly by removing the taxels that aren't transmitted.

moreover I think that change wrapper's configuartion is not easy, so I can improve the documentation asap.

sorry for typo

closed 2021-03-27 by pattacini