robotology/icub-main #532: ICUB devel is not configuring with latest YARP devel

traversaro, 2018-06-20:
Configuring latest ICUB devel  with latest YARP devel, there are a lot of CMake errors like:

~~~
CMake Error at CMakeLists.txt:98 (include):
  include could not find load file:

    YarpInstallationHelpers

-- Setting up installation of iCub.ini to /home/straversaro/src/robotology-superbuild/build/install/share/yarp/config/path.d folder.
CMake Error at src/libraries/icubmod/CMakeLists.txt:8 (include):
  include could not find load file:

    YarpPlugin
~~~

Reverting this commit: https://github.com/robotology/icub-main/commit/fe10cefe4d19958d8d7618d7a3867205fce2980d seems to fix the problem. 

@pattacini @drdanz

traversaro, 2018-06-20:
I suspect the warning: 
~~~
CMake Deprecation Warning at /home/straversaro/src/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:272 (message):
  The YARP_MODULE_PATH variable is deprecated.  CMake find package modules
  are now in YCM.
~~~ 
is a bit misleading.

pattacini, 2018-06-21:
Yes indeed, thanks @traversaro I'll deal with the revert.
By the way, I did a bunch of quick tests on other repos and all went well; that's why I extended to icub-main too.

@drdanz what I have to do to fix the warning?

pattacini, 2018-06-21:
Commit removed in the meantime.

pattacini, 2018-06-21:
References to `YARP_MODULE_PATH` can be safely removed and, at the same time, we ought to remove directives such as `include(YarpInstallationHelpers)` or `include(YarpIDL)`, which in turn are already automatically included by YARP.

To get `ACE` instead, one needs to rely on `YCM`.

closed 2018-06-21 by pattacini