robotology/icub-main #534: FindACE w/o YARP_MODULE_PATH

pattacini, 2018-06-26:
Since `YARP_MODULE_PATH` is deprecated, to find `ACE` we would need to rely on `YCM`, meaning that `YCM` should be considered from now on a new strict dependency in our pipeline, also affecting the content we deploy through our binaries.

cc @traversaro @drdanz @lornat75 @mbrunettini @vtikha @claudiofantacci

claudiofantacci, 2018-06-26:
Since this action may require some extra effort, I'm wondering if it is worth setting up a f2f meeting to opt for proper actions.

Tobias-Fischer, 2018-12-10:
Is there any news for that issue? Sooner or later people will run into issues, for example see https://github.com/robotology/yarp/issues/1831 which is closed but not fixed ..

drdanz, 2018-12-11:
> Don't know when we've planned to consider YCM a strict dependency.

About 4 years ago...

pattacini, 2018-12-11:
@drdanz the question was related to our packages, I suppose 😉

pattacini, 2018-12-11:
I wouldn't put YCM in `icub-main` honestly, it's quite a dirty solution.
Let's deliver it in the new packages and, in the meanwhile, if YCM is required travis-wise, then it could be still downloaded in the server easily.

drdanz, 2018-12-11:
> I wouldn't put YCM in icub-main honestly, it's quite a dirty solution.

That's the reason why it's not there yet :rofl:

pattacini, 2020-03-02:
`ycm` is mandatory for `icub-main` sincce distro 2020.02.

closed 2020-03-02 by pattacini