robotology/icub-main #535: Inconsistency between PWM calculated, and PWM measured in iCub joints

vibhoraggarwal, 2018-07-03:
**Description:** 
I did some experiment with the PWM on `l_hip_pitch` and `l_knee`, and PWM was recorded as follows:
- $ \textrm{Calculated PWM}= K_\tau \ast (K^{ff}\tau^d+K_{bemf}*\textrm{(Gearbox_M2J)}\ast\dot\theta)+\Delta PWM_{PID} $
- **Measured PWM** is fed from the WBToolbox's `Get Motor Measurement`
- In the experiment, $\Delta PWM_{PID} = K_p \ast (\tau^d-\tau^{est})$ for simplicity

It was **observed**:
- In `l_hip_pitch`: Calculated and Measured PWM are equal only when $K_p =0$. As soon as the value of $K_p \neq 0$, PWM from the feedforward terms are neglected
- In `l_knee`: Feedforward PWM are never considered, and are assumed zero in the measurement

This is a bit surprising because the PWM in firmware is calculated using [Motor_do_trq_control()](https://github.com/robotology/icub-firmware/blob/084590ccd280c8808ce0ecd215f45f27c0ffc279/emBODY/eBcode/arch-arm/embobj/plus/mc/Motor.c#L798 ), whose definition is consistent with above
cc: @traversaro

vibhoraggarwal, 2018-07-03:
**Joint**: `l_hip_pitch`:
1. As soon as the **feedback term is non zero**, measured PWM  simply avoids the PWM generated using the feed forward term, and measured PWM is wrong
Following proves this: 
   - $K_p=0, K_\tau K_{bemf} \neq 0, $
[image]
   - $K_p \neq0$, to be precise $K_p=10^{-30}$, $K_\tau K_{bemf} \neq 0, $
[image]
2. It is to be noticed that the PWM measured with $K_p \neq 0$, does measure the PWM generated from the **feedback** terms:
[image]

vibhoraggarwal, 2018-07-03:
The behavior is a bit different in case of `l_knee`: **The feed forward PWM is never considered.**

Following is the PWM plot in case of $K_p=0, K_\tau K_{bemf} \neq 0, $[image]

traversaro, 2018-07-03:
@vibhoraggarwal Can you provide information of the iCub robot on which you experienced this problem? 

@pattacini I asked @vibhoraggarwal to open the issue here even if it is seems to be something firmware-related because it is a iCub-specific issue.

cc @DanielePucci

vibhoraggarwal, 2018-07-03:
@traversaro 
>  information of the iCub robot 

[Description](https://github.com/robotology/icub-main/issues/535#issue-337962394) updated

traversaro, 2018-07-03:
It would be good to try to replicate this findings on `iCubGenova04`, when it is available.

traversaro, 2018-07-03:
@ale-git @marcoaccame @valegagge any idea?

DanielePucci, 2018-07-03:
This is something of really high priority since if it is confirmed, I sincerely wonder how the torque control related demos work

CC
@randaz81

traversaro, 2018-07-03:
@vibhoraggarwal forgot to mention that while he set all the parameters through the [WB-Toolbox](https://github.com/robotology/wb-toolbox), we always checked that the values were set to the correct values using the `yarpmotorgui`.

vibhoraggarwal, 2018-07-04:
**Update:**
I again checked the PWM by:
- Setting the parameter through `yarpmotorgui` and **not [WBToolbox's ](https://github.com/robotology/wb-toolbox)** `Set Motor Parameters` block
- But, still, reading the PWM through WBToolbox

**Observation:** 
- The PWM reading are matching in the correct way; **saving us from going through all the firmware code** @DanielePucci @traversaro 
- Then it seems that there is a problem in the way parameters are being set in WBToolbox; but, which is **unlikely**, as the change is being reflected in the `yarpmotorgui`. @diegoferigo could comment on the behaviour in a much better way
- Thus, the problem could be changing the **parameters for a small duration of time**, which is also the reason why we did not check it by changing through `yarpmotorgui`, as the change in the parameters done thorugh WBToolbox's `Set Motor Parameters`were being reflected in `yarpmotorgui`

ale-git, 2018-07-04:
@vibhoraggarwal Actually I was going through the firmware finding nothing weird :)
Please consider that some robots are using FW from the soft_pid_config branch, where the parameter transition in PID controllers is smooth, with user defined rate of change.

traversaro, 2018-07-04:
>  Thus, the problem could be changing the **parameters for a small duration of time**, which is also the reason why we did not check it by changing through `yarpmotorgui`, as the change in the parameters done thorugh WBToolbox's `Set Motor Parameters`were being reflected in `yarpmotorgui

This may be related to the use of the `soft_pid_config` branches/firmware that @ale-git mentioned.

diegoferigo, 2018-07-04:
As a reference, the file [`SetMotorParameters.cpp`](https://github.com/robotology/wb-toolbox/blob/devel/toolbox/src/SetMotorParameters.cpp) is the responsible of setting the PIDs, Ktau and BEMF constant.

traversaro, 2018-09-06:
Hi @vibhoraggarwal , can you update this issue when you have time? Thanks!

traversaro, 2018-09-21:
Not sure, but I think @vibhoraggarwal eventually understood the issue. Closing the issue, feel free to re-open if necessary.

closed 2018-09-21 by traversaro

vibhoraggarwal, 2018-09-21:
Hi @traversaro; Sorry,I somehow missed the notification. 
I would like to point out, that there was no problem with the firmware; I checked the same thing in `iCubGenova04`, and the calculated and measured PWM were consistent. 
Although the exact issue in `iCubGenova02` was never rectified, because of problems like lack of updated software in the robot,etc. It may be a different issue based on hardware.