robotology/icub-main #536: icub-main can not "CMAKE" anymore with YARP 3.0.0

caicaixia, 2018-07-04:
Hello guys,
Today I am installing the yarp and icub-main for my new computer.  Then I found out that the yarp have the new verson: YARP3.0.0;

1: I installed the YARP 3.0.0;
2:  git clone https://github.com/robotology/icub-main.git
However, I can not CMAKE because of the new version of the YARP.
[image]

Q: HOW can I solve this problem?
     Should I reinstall the old version of yarp,  or should I install the "icub-main" devel branch?

What is the recommendation from you for the new installation?

pattacini, 2018-07-04:
Hi @caicaixia 

Because of https://github.com/robotology/QA/issues/300, the suggestion is to switch to `devel` branches, both for `yarp` and `icub-main`.

Alternatively, you can `git checkout` the tags corresponding to our releases. To know the correct release pairs, visit http://wiki.icub.org/wiki/Downloads.

caicaixia, 2018-07-04:
ok.  Thanks. I will use the old yarp for now :+1:

closed 2018-07-04 by pattacini