robotology/icub-main #552: Error in linking the CAN libraries in Windows

claudia-lat, 2018-09-13:
I've tried to compile icub-main 
- on a Windows 10 laptop 
- with a USB-CAN connected to it
- be enabling the following options: ```ENABLE_icubmod_ecan```, ``` ENABLE_icubmod_canBusAnalogSensor```.

I got an error in the libraries linking:

```bash
 "C:\Users\icub\devel\icub-main\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "C:\Users\icub\devel\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj" (default target) (17) ->
       (Link target) ->
         EsdCan.obj : error LNK2019: unresolved external symbol canSetBaudrate referenced in function "public: virtual bool __cdecl yarp
       ::dev::EsdCan::canSetBaudRate(unsigned int)" (?canSetBaudRate@EsdCan@dev@yarp@@UEAA_NI@Z) [C:\Users\icub\devel\icub-main\build\sr
       c\libraries\icubmod\esdCan\esdCan.vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canGetBaudrate referenced in function "public: virtual bool __cdecl yarp
       ::dev::EsdCan::canGetBaudRate(unsigned int *)" (?canGetBaudRate@EsdCan@dev@yarp@@UEAA_NPEAI@Z) [C:\Users\icub\devel\icub-main\bui
       ld\src\libraries\icubmod\esdCan\esdCan.vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canOpen referenced in function "public: virtual bool __cdecl yarp::dev::
       EsdCan::open(class yarp::os::Searchable &)" (?open@EsdCan@dev@yarp@@UEAA_NAEAVSearchable@os@3@@Z) [C:\Users\icub\devel\icub-main\
       build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canClose referenced in function "public: virtual bool __cdecl yarp::dev:
       :EsdCan::close(void)" (?close@EsdCan@dev@yarp@@UEAA_NXZ) [C:\Users\icub\devel\icub-main\build\src\libraries\icubmod\esdCan\esdCan
       .vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canIdAdd referenced in function "public: virtual bool __cdecl yarp::dev:
       :EsdCan::canIdAdd(unsigned int)" (?canIdAdd@EsdCan@dev@yarp@@UEAA_NI@Z) [C:\Users\icub\devel\icub-main\build\src\libraries\icubmo
       d\esdCan\esdCan.vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canIdDelete referenced in function "public: virtual bool __cdecl yarp::d
       ev::EsdCan::canIdDelete(unsigned int)" (?canIdDelete@EsdCan@dev@yarp@@UEAA_NI@Z) [C:\Users\icub\devel\icub-main\build\src\librari
       es\icubmod\esdCan\esdCan.vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canTake referenced in function "public: virtual bool __cdecl yarp::dev::
       EsdCan::canRead(class yarp::dev::CanBuffer &,unsigned int,unsigned int *,bool)" (?canRead@EsdCan@dev@yarp@@UEAA_NAEAVCanBuffer@23
       @IPEAI_N@Z) [C:\Users\icub\devel\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canRead referenced in function "public: virtual bool __cdecl yarp::dev::
       EsdCan::canRead(class yarp::dev::CanBuffer &,unsigned int,unsigned int *,bool)" (?canRead@EsdCan@dev@yarp@@UEAA_NAEAVCanBuffer@23
       @IPEAI_N@Z) [C:\Users\icub\devel\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
         EsdCan.obj : error LNK2019: unresolved external symbol canWrite referenced in function "public: virtual bool __cdecl yarp::dev:
       :EsdCan::canWrite(class yarp::dev::CanBuffer const &,unsigned int,unsigned int *,bool)" (?canWrite@EsdCan@dev@yarp@@UEAA_NAEBVCan
       Buffer@23@IPEAI_N@Z) [C:\Users\icub\devel\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
         C:\Users\icub\devel\icub-main\build\lib\iCub\Release\esdCan.dll : fatal error LNK1120: 9 unresolved externals [C:\Users\icub\de
       vel\icub-main\build\src\libraries\icubmod\esdCan\esdCan.vcxproj]
```

After several tests with @marcoaccame, we discovered that if you manually modify the options
```ESDCANAPI_LIB``` by changing the library path from  ```C:/ProgramFiles/ESD/CAN/SDK/lib/VC/i368/ntcan.lib``` to ```C:/ProgramFiles/ESD/CAN/SDK/lib/VC/amd64/ntcan.lib```, the linking works and we are able to compile. 
 Is the problem located [here](https://github.com/robotology/icub-main/blob/master/conf/FindESDCANAPI.cmake)?

Yeshasvitvs, 2018-11-08:
The option `ESDCANAPI_LIB` can only be seen by ticking the `Advanced` option in cmake-gui!

@claudia-lat

traversaro, 2018-11-08:
I guess the `C:/ProgramFiles/ESD/CAN/SDK/lib/VC/i368/ntcan.lib` is ok when linking 32-bit binaries, while  `C:/ProgramFiles/ESD/CAN/SDK/lib/VC/amd64/ntcan.lib` should be used when linking 64-bit binaries. 
I guess it would make sense to have a code like (see https://stackoverflow.com/questions/39258250/how-to-detect-if-64-bit-msvc-with-cmake) : 
~~~
if(CMAKE_SIZEOF_VOID_P EQUAL 8)
  # 64-bit    
  set(ESDCANAPI_ARCH "amd64")
else()
  # 32-bit 
  set(ESDCANAPI_ARCH "i368")
endif()
~~~
and then use appropriately the `${ESDCANAPI_ARCH}` in the directories that are appended. 
@InesSorrentino can you address this? Thanks!

traversaro, 2018-11-08:
Note that the actually file to modify is https://github.com/robotology/ycm/blob/7cc2b755e302b972112ca6b77a6dbce4492d0062/find-modules/FindESDCANAPI.cmake .

traversaro, 2018-11-08:
Related use of `CMAKE_SIZEOF_VOID_P` : https://github.com/robotology/ycm/blob/811edb419ab9731a8bfead99b6361cc4f0f3f595/find-modules/FindGLUT.cmake#L49 .

Yeshasvitvs, 2018-11-08:
Thanks @traversaro

InesSorrentino, 2018-11-08:
@traversaro yes, I'll fix it.