robotology/icub-main #562: FirmwareUpdater libGL errors with iCub LIVE image 7.9

gsaponaro, 2018-12-16:
We recently installed the iCub LIVE image 7.9 on a new USB pen drive for the PC104 of iCubLisboa01 (CAN robot), and we updated the software in the robot cluster (yarp 3.1.0 from master).

The robot is operational, we configured our network, we tested the red ball demo successfully.

However, when we launch the `FirmwareUpdater` tool (with or without options) it complains about the following:

```
$ ssh -X pc104
iCub LIVE image for PC104 version 7.9-17.12.18
Based on stretch amd64
Build on 18-Dec-2017
Linux pc104 4.9.0-4-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.65-3 (2017-12-03) x86_64
[...]
icub@pc104:~$ FirmwareUpdater

ACE_SOCK_Dgram: Cannot assign requested address
ACE_SOCK_Dgram_Bcast::mk_broadcast: setsockopt failed: Operation not supported
ACE_SOCK_Dgram_Bcast: Operation not supported
"CFW2" - "0"
"CFW2" - "1"
"CFW2" - "2"
"CFW2" - "3"
"CFW2" - "4"
"CFW2" - "5"
"CFW2" - "6"
"CFW2" - "7"
"CFW2" - "8"
"CFW2" - "9"
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
```

[image]

@marcoaccame , could you help us in getting `FirmwareUpdater` to run? We seem to be missing some Debian packages, and probably a configuration file.

julijenv, 2018-12-18:
hi @gsaponaro , sorry, my bad I sent you the wrong file. here is the good one, please try again.

[calibrationDataSN094.txt](https://github.com/robotology/icub-main/files/2690475/calibrationDataSN094.txt)
in order to send it I had to change the extension form dat to txt please rename it in .dat

gsaponaro, 2018-12-18:
Hi @julijenv , thank you very much for that: I updated the other issue https://github.com/robotology/icub-support/issues/749 after the now-successful F/T flashing.

I keep this issue open (https://github.com/robotology/icub-main/issues/562) about the software problem that we encountered with `FirmwareUpdater`.

julijenv, 2018-12-18:
to @marcoaccame , i was toalking about that issue, thx in advance to have a look at it

marcoaccame, 2019-01-02:
The FirmwareUpdater program seems OK to me. It loads the correct drivers CFW2<x> as indicated by its `firmwareupdater.ini`.

I don't know about the libGL and ACE_SOCK_Dgram messages, but I would let them aside for a while.

What does happen if you select (put the mouse onto and click) the CFW2<2> device and you press `Discover`? 

The effect should be to launch a discovery on the device using the same library as the canloader does as for example in https://github.com/robotology/icub-support/issues/749 and i expect to see a strain, two blls and one mc4.

gsaponaro, 2019-01-03:
Hi @marcoaccame , thanks for the help.

> What does happen if you select (put the mouse onto and click) the CFW2<2> device and you press `Discover`?

It says that no CAN boards are found, screenshot:

[image]

Log:

```bash
icub@pc104:~$ FirmwareUpdater --verbosity 1
Using default address  10.0.1.104
Using default port  3333
||| clearing context
||| adding context [firmwareUpdater]
||| configuring
||| default config file specified as firmwareupdater.ini
||| checking [/home/icub/firmwareupdater.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubLisboa01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubLisboa01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubLisboa01
||| checking [/etc/yarp/robots/iCubLisboa01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/robots/iCubLisboa01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubLisboa01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubLisboa01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-tests/suits/robots/iCubLisboa01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-tests/suits/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubLisboa01/firmwareupdater.ini] (robot)
||| found /home/icub/.local/share/yarp/robots/iCubLisboa01/firmwareupdater.ini
ACE_SOCK_Dgram: Cannot assign requested address
ACE_SOCK_Dgram_Bcast::mk_broadcast: setsockopt failed: Operation not supported
ACE_SOCK_Dgram_Bcast: Operation not supported
"CFW2" - "0"
"CFW2" - "1"
"CFW2" - "2"
"CFW2" - "3"
"CFW2" - "4"
"CFW2" - "5"
"CFW2" - "6"
"CFW2" - "7"
"CFW2" - "8"
"CFW2" - "9"
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
[INFO]created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
FirmwareUpdaterCore::getCanBoardsFromDriver(): No answer received from CAN boards after a successful driver init.
[INFO]created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
FirmwareUpdaterCore::getCanBoardsFromDriver(): No answer received from CAN boards after a successful driver init.
```

By the way, our robot is CAN, and our PC104 has an IP address different than the default: 10.10.1.50. Shall we make a custom `firmwareupdater.ini`? If so, can you point me to any examples of it?

gsaponaro, 2019-01-03:
Sorry, please disregard the last message. I can actually `Discover` the two CAN nets of the arms:

[image]

The result seems consistent to what I was getting with `canLoader` in https://github.com/robotology/icub-support/issues/749. I will now write in the other issue the result of `FirmwareUpdater --admin` calibration window.

closed 2021-03-27 by pattacini