robotology/icub-main #563: Investigate esdCan or canBusAnalogSensor driver termination

Yeshasvitvs, 2018-12-19:
While using the FTShoes we noticed immediately after killing `yarprobotinterface` with `ctrl+c` I could not relaunch it again. It almost always throws the error shown below and the I need to remove and plugin the usb of the shoe cables before launching `yarprobotinterface` again

[image]

Check this issue for more details related to the problem https://github.com/loc2/element_ftSk_Shoes/issues/95#issue-388738747

Ideally, the termination should have been smoother and the FT sensors should be accessed by launching yarprobotinterface without any loss in connection or other errors. We suspect that there is a possible bug in either esdCan or canBusAnalogSensor driver. 

@traversaro @InesSorrentino @fjandrad @claudia-lat @DanielePucci

traversaro, 2018-12-19:
I think it is worth mentioning this addition info:
* The FTShoe is a sandal with two IIT FTSens sensor mounted on the same CAN network.
* Apparently the issue does not arise when only one FT sensor is connected to the CAN network (@InesSorrentino correct me if I am wrong) 
* The same issues arises on both Linux (using socketCan) and Windows (using esdCan) (@InesSorrentino please correct me), so we suspect that the issue is that `canBusAnalogSensor` is not handling correctly shuting down multiple sensors (even if it used to work fine on the legs of `iCubGenova01` .
 

> Check this issue for more details related to the problem loc2/element_ftSk_Shoes#95 (comment)

I don't think it is useful to link private repo issues in a public repository. Feel free to copy all the relevant information here.

InesSorrentino, 2018-12-19:
@traversaro everything is correct.

traversaro, 2018-12-19:
This issue could also be related to the STRAIN2 firmware. @marcoaccame @valegagge

valegagge, 2018-12-21:
As @traversaro said, I think the problem is due to how the canBusAnalogSensor deinitialises the strain before closeing itsself; in order to understand if the bug is in canBusAnalogSensor or in the sensor it would be great if we have the following information:

1.  The setup of shoes used by @Yeshasvitvs is composed by Strain1, right? 
2. Does exist a setup (maybe on green robot) where 2 Strain2 are connected to same can bus of an Ethernet board? If exists, do someone has experience of this bug?
3. Does exist a setup (maybe on green robot) where 2 Strain2 are connected to same can bus and read by canBusAnalogSensor?  If exists, do someone has experience of this bug?
4. @Yeshasvitvs : did you do ctrl+c one time only? 

> ...we suspect that the issue is that canBusAnalogSensor is not handling correctly shuting down multiple sensors (even if it used to work fine on the legs of iCubGenova01

Maybe the configuration of shoes and on leg of Genova01 aren't exactly the same....we need to check this .

Yeshasvitvs, 2018-12-21:
> As @traversaro said, I think the problem is due to how the canBusAnalogSensor deinitialises the strain before closeing itsself; in order to understand if the bug is in canBusAnalogSensor or in the [...]

1: I am using Pro.04 FTShoes which contain Strain 2 FTSensors
2, 3:  I am not an active user and I am not aware of someone who is. Probably @fjandrad can give some inputs on this.
4. I have a habit of doing it 3 times and I cannot be certain of exactly how many times I was using ctlr+c to kill the yarprobotinterface for FTShoes 

@valegagge

traversaro, 2018-12-21:
> 1: I am using Pro.04 FTShoes which consists of Strain 2 FTSensors

I can't find the relevant yarprobotinterface conf file for pro04 in https://github.com/robotology-playground/forcetorque-yarp-devices/tree/master/ftShoe/conf . Where it is?

Yeshasvitvs, 2018-12-21:
> > 1: I am using Pro.04 FTShoes which consists of Strain 2 FTSensors
> 
> I can't find the relevant yarprobotinterface conf file for pro04 in https://github.com/robotology-playground/forcetorque-yarp-devices/tree/master/ftShoe/conf . Where it is?

In general we push the config file once we do the calibration with the force platform and find the inSituMatrices, for example check Pro.02 [inSituMatrices](https://github.com/robotology-playground/forcetorque-yarp-devices/blob/7c08e4fd78e705a3eb88b8315b143815a3709a8c/ftShoe/conf/ftshoes_yarprobotinterface_PRO.02.xml#L81). In the case of Pro.04 we did not do the calibration yet as the platform was removed by then from morego due to the move. So, I am simply using (locally) the Pro.03 config file with [useInSituCalibration](https://github.com/robotology-playground/forcetorque-yarp-devices/blob/7c08e4fd78e705a3eb88b8315b143815a3709a8c/ftShoe/conf/ftshoes_yarprobotinterface_PRO.02.xml#L79) flag set to **false**.

traversaro, 2018-12-21:
> So, I am simply using (locally) the Pro.03 config file with useInSituCalibration flag set to false.

For @valegagge : the PRO.03 config is https://github.com/robotology-playground/forcetorque-yarp-devices/blob/master/ftShoe/conf/ftshoes_yarprobotinterface_PRO.03.xml .

fjandrad, 2018-12-21:
@valegagge on the green each sensor has a different address so I would assume they are in a different Ethernet board. And since the green uses Ethernet I doubt it has a can configuration with 2 sensors, with the shoes we connect two per feet so sometimes there are up to 4 in the same setup.

valegagge, 2018-12-21:
@Yeshasvitvs : if you use more than one ctrl+c , maybe you can abort some operation, so if you want to shutdown the setup in a clean way, please press one time only.  In this way all deinitialize procedures has time to complete their works and leave the board in a correct state.

valegagge, 2018-12-21:
@fjandrad 

> on the green each sensor has a different address so I would assume they are in a different Ethernet board.
This is not complete correct, each Ethernet board could have more one can boards connected (like mtb), but I saw, in green configuration file on devel branch, that each strain has the same CAN address, so this mean that there is one strain per CAN bus. Moreover I checked that there is one strain per Ethernet board.

I'm sorry, in the question 3 I did an error: I would to know if someone use 2 strain 2 on same Ethernet board, but I think this setup doesn't exist. So forget question 3, please :)

valegagge, 2018-12-21:
In the configuration of shoes setup I noticed the device called ftshoe. What is this?

However to solve this issue, first of all I would like to use only one crtl+c.
If the problem happens again, I can investigate it, but I need a your setup with the two or four strain connected together.

traversaro, 2018-12-21:
> In the configuration of shoes setup I noticed the device called ftshoe. What is this? 

* https://github.com/robotology-playground/forcetorque-yarp-devices/blob/master/ftShoe/ftshoeDriver.h
* https://github.com/robotology-playground/forcetorque-yarp-devices/blob/master/ftShoe/ftshoeDriver.cpp 

In short, it is a  device that attached to the two FT sensors and sums their measure to expose them as a single "combined" FT sensor.

Yeshasvitvs, 2018-12-21:
> In the configuration of shoes setup I noticed the device called ftshoe. What is this?
> 
> However to solve this issue, first of all I would like to use only one crtl+c.
> If the problem happens again, I can investigate it, but I need a your setup with the two or four strain connected together.

May be @InesSorrentino tried to kill it with one ctrl+c ?

InesSorrentino, 2019-01-11:
Yesterday I tried the behavior of the ctrl + c to stop the drivers of the ftsensors (I'm running the driver of the right shoe and that one of the left shoe together but on two separate terminals).
If you stop the driver by typing repeatedly and quickly ctrl + c the problem happens.
Instead the correct way is the following. If after one ctrl+c the driver is not stopped you can try again but by waiting a bit of time between the first and the second ctrl+c. In this case (even in the case you type more then one or two ctrl+c) all works fine at the restarting. The important thing is to wait before type a further ctrl+c.

I think that this issue can be closed.

cc @Yeshasvitvs @valegagge

traversaro, 2019-01-11:
I think a single Ctrl+C should be sufficient. Is this the case @InesSorrentino ? Otherwise, there are still some termination issues.

InesSorrentino, 2019-01-11:
It depends, in most cases with a single Ctrl+C the driver is not stopped. Thus, if the correct way must be with a single Ctrl+C, then probably there are still issues.

traversaro, 2019-01-11:
Let's wait @valegagge reply, but I think a single Ctrl+C should be sufficient.

DanielePucci, 2019-01-20:
CC
@valegagge

valegagge, 2019-01-21:
A single Ctrl+C is sufficient to shutdown yarprobotinterface. In some cases, the user need to wait more than 30 seconds before the shutting down phase is finished.
Let me know if there are still issues.

closed 2021-03-27 by pattacini