robotology/icub-main #568: Motor current measurement excessive oscillations and dependency on motor angular position

nunoguedelha, 2019-02-12:
On a slow moving (*) joint and a **constant PWM set-point**, the motor current measurement displays a random noise of $300 mA$ about a mean value which is not constant as we expected, but varying with the motor position. The amplitude of the current oscillations is of +/-0.5 A.

(*) The joint is being move slowly to static positions and kept a few seconds on each position.

nunoguedelha, 2019-02-15:
The tests were performed on the hip roll joint of the left leg of iCub.

### Fixed PWM set-point $\rightarrow$ expected constant measured current
While for a fixed PWM set-point the mean current value is expected to be constant, we observe variations within $\pm 0.5 A$ about a fixed value.

In a first experiment, we set and keep the PWM at a fixed value for a given period of time, through three consecutive phases:
- PWM = 0 % duty-cycle for ~ 50 s,
- PWM = 4 % duty-cycle for ~ 90 s,
- PWM = 0 % duty-cycle for ~ 100 s.
In figure 1 we plot a the PWM and motor current timeseries, and in figure 2 the current against the PWM.

[image]
Figure 1.

[image]
Figure2.

During each phase, we alternatively let the joint move for a brief moment onto a new position before blocking it again and keeping it still for longer periods (between 10s and 20s). Let us take a close look at the period 57s -> 144s:
As the $k_ {\mathrm{pwm},i} \sim 0.5 A/\mathrm{dutycycle\\%}$, we would expect a current always $\sim 2 A$. Intsead we measure a different current for each different position, and a noise around 300mA:
- 58s to 86s => 2.2 A
- 88s to 98s => 0.89 A
- 123s to 134s => 1.9 A

Same oscillations are observed with PWM = 0 dutycycle%. For the period 7s to 54s the motor position is fixed. In figure 2 we can see that for a fixed PWM, the current spans a set of values between -1.2 A and 1.2 A.

In a second experiment the PWM is set following a triangular waveform, with 0.3 Hz frequency, while blocking the leg. In spite of the position constraint, the leg structure and harmonic drive flexibility allow the motor to rotate over a around 3 turns.

nunoguedelha, 2019-02-15:
CC @lrapetti  @DanielePucci   @traversaro

nunoguedelha, 2019-02-15:
CC @maggia80

lrapetti, 2019-02-15:
> that offset depends on the rotor position and does not depend on the PWM.

@nunoguedelha
Did you checked if, for a given rotor position the offset is constant in time and/or in repeated tests (e.g. afternoon turning off the robot).

nunoguedelha, 2019-02-15:
The current is constant for a given fixed rotor position, at least in a period on a couple of minutes. 
Any hypothetical drift over a longer period would have nothing to do with what we are observing here I think.

**Note:** Regarding any experiment where we turn off and back on the robot, we have to keep in mind that the motor position is given by relative encoders so we won't have the same readings for a given position after a power cycle. So after a power cycle, for a given joint absolute position, we will have a different motor current for sure. The gearbox aggravates this aspect since 1deg of joint position implies 100deg of motor rotation, so...

We can do a test to see if there is any kind of drift in the current measurement with the robot ON. But this is another matter..

DanielePucci, 2019-02-16:
CC
@ale-git @maggia80

nunoguedelha, 2019-02-16:
Updated comment https://github.com/robotology/icub-main/issues/568#issuecomment-464099790 with a GIF clarifying the projection (Figure 4b).

maggia80, 2019-02-22:
we should look at how the current is measured. Remember that for low PWM duty cycle the measurement of the current is quite noisy. This is because usually, you measure the current when the PWM is in the ON state and then you divide it for the percentage of the PWM (in order to get the mean of the current during the whole PWM period). But I don't know how @ale-git is doing the measurement in the FW. The one I have described is one option, but there are many others.

nunoguedelha, 2019-02-24:
@maggia80 I see your point. Knowing how the current is measured should help us understand the $\pm 100 mA$ noise.
Still, I would like to highlight the following points to consider regarding the problem of $\pm 0.5 A$ oscillations, i.e. the observed offset $\Delta i = f(\theta_m)$, where $\theta_m$ is the rotor position:
- Considering that we get a pattern clearly depending on the position, I believe we all agree that we are not facing a noise problem,
- it is not related with the back EMF since we observe the same pattern for different **static** positions,
- the pattern has an amplitude of 0.5 A, whatever the PWM value.

As @ale-git explained me, the "measured" current is an estimate of the current quadrature axis component $i_q$ in the commonly called direct-quadrature frame (d-q). In that case, it would be worth checking the Park transform, in other words the overall computation:
$$
i_q = f(i _{phaseA},i _{phaseB},i _{phaseC},\theta_m)
$$

**Note**
It is worth checking the current measurement while the motor is in **idle mode**, for verifying that the oscillations are not caused by the inverse Park transform projecting the input PWM from the d-q frame to the fixed stator frame.

nunoguedelha, 2019-02-24:
Edited my last comment.

maggia80, 2019-02-26:
I see

nunoguedelha, 2019-03-12:
Ciao @ale-git, would you have any news on this issue?

If I'm not mistaken @DanielePucci and @maggia80 considered the possibility that we work together on testing the current control on the iCub2.5 (iCubGenova04). In that case I would be quite interested in taking a look at the firmware code handling the Clark & Park transforms, for having an idea of how it all works.

nunoguedelha, 2019-04-01:
@ale-git , I've fixed the bug preventing me to show you the experiment. Btw, here is a new plot, simpler, showing the current response for a constant PWM input = 6 % while I hold the leg, but let it move very slowly. The moving joint is the left hip roll.

[image]

I will be busy until next Monday 8 april end of day preparing the Defense.

See you on Tuesday then!

ale-git, 2019-04-02:
@nunoguedelha Thanks Nuno, I'm checking the current computation starting from the cabling diagram up to the firmware.

nunoguedelha, 2019-04-15:
ciao @ale-git , any news?

ale-git, 2019-04-15:
Hi @nunoguedelha, probably the gain of the two current sensors are slightly unbalanced. There is a calibration procedure at startup for the offset, so I'm thinking about extending it to the calibration of the gain.

closed 2023-07-30 by pattacini