robotology/icub-main #569: icubSim: continuous crashes due to ODE INTERNAL ERROR 1

valeriosperati, 2019-03-05:
Greetings,
about icubSim, I continuously report crashes after few minutes, during which iCub moves using both Cartesian Interface and Gaze Interface. I always obtain the same error:  

ODE INTERNAL ERROR 1: assertion "(g1->gflags & GEOM_AABB_BAD)==0" failed in collideAABBs() [collision_space_internal.h:51]

Note that in my simulation the iCub interacts with simulated standard objects (i.e. cubes and spheres). Nevertheless iCubSim crashes even if robot stands still, generally after a couple of minutes. 

Any idea?
Thanks, Valerio

System: Ubuntu 16.04, icub-main compiled from sources.

pattacini, 2019-03-05:
Is it also crashing with iCub standing still and with or without objects?

Ideally, it would be beneficial to be able to reproduce your environment in the least amount of things to make the crash occur.

valeriosperati, 2019-03-06:
Hi, thanks for the suggestion, it helped me in solving the problem.
I realised that the problem arises when I work with objects in the environment. 
I checked my code and I noticed that I erroneously called continuously (at each step of the simulation) the command 
`world set ssph`
 on a sphere which was supposed to be still. Maybe this useless, continuous command generated the ODE error. 
Removing this command solved the problem, now the simulation seems to be stable. I'll run several tests.
Thanks again.

closed 2019-03-06 by valeriosperati

pattacini, 2019-03-06:
Happy to have been of any help 👍