robotology/icub-main #571: iCub::iKin::iCubFinger::getChainJoints uses the wrong joint limits for the thumb

xEnVrE, 2019-03-26:
The [method](https://github.com/robotology/icub-main/blob/e030379c948d347f96a942058c4f5269234a4116/src/libraries/iKin/src/iKinFwd.cpp#L2145) `iCub::iKin::iCubFinger::getChainJoints(const Vector &motorEncoders, const Vector &jointEncoders, Vector &chainJoints, const Matrix &jointEncodersBounds)` uses the operator `iKinLink&  iCub::iKin::iKinChain::operator[]` in order to obtain the limits of the finger joints. 

Specifically, the maximum is taken as 
`(*this)[i].getMax()` while the minimum as `(*this)[i].getMin()` (see the code below).

https://github.com/robotology/icub-main/blob/e030379c948d347f96a942058c4f5269234a4116/src/libraries/iKin/src/iKinFwd.cpp#L2164-L2178

Unfortunately, the `thumb` is different from other fingers because, when data is allocated in `iCub::iKin::iCubFinger::allocate()`, one of the link is blocked since it is a dummy one

https://github.com/robotology/icub-main/blob/e030379c948d347f96a942058c4f5269234a4116/src/libraries/iKin/src/iKinFwd.cpp#L1910-L1913

Since the `operator[]` does not skip dummy links, when the index `i` assumes the value `i = 1`, in the `for` loop indicated above, the link returned by `(*this)[1]` is the dummy link instead of being the link associated to the proximal joint of the thumb as expected. For the dummy link the difference `(*this)[i].getMax()-(*this)[i].getMin()` is always zero, since the limits associated to it are both zero, resulting in a value `chainJoints[1]` that is always zero independently of the analog readings for the thumb proximal joint.

Should we use `iKinLink&  iCub::iKin::iKinChain::operator()` that, according to the [documentation](http://www.icub.org/software_documentation/classiCub_1_1iKin_1_1iKinChain.html#ae2eed7a30de062e961c5c1bc09ab4653), consider only links associated to effective DOFs of the chain, i.e. non blocked links?

@pattacini

pattacini, 2019-03-26:
Good catch @xEnVrE 👍 
Should be fixed in `devel` now.

I've never played in depth with the fingers, thus it might be that there are other similar mistakes.

Do you think/expect that this patch might also help narrow down the longstanding problem we've been observing with the thumb?

cc @claudiofantacci

xEnVrE, 2019-03-27:
> Should be fixed in devel now.

Thanks @pattacini for the fix.

> Do you think/expect that this patch might also help narrow down the longstanding problem we've been observing with the thumb?

The main problem we have have the thumb affects the **opposition** joint while this bug was mainly affecting the  **proximal** joint.

pattacini, 2019-03-27:
All clear, closing then.

commit 465be67d (2019-03-26): [iKin] fix #571
closed 2019-03-27 by pattacini