robotology/icub-main #577: FirmwareUpdater does not build on latest icub-main devel with latest YARP devel on Windows

InesSorrentino, 2019-05-09:
~~~

"C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\FirmwareUpdater\FirmwareUpdater-qt.vcxproj" (default target) (3) ->
(Link target) ->
  canLoaderLib.lib(driver.obj) : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual class yarp::dev::CanBuffer __cdecl yarp::dev::ImplementCanBufferFactory<class EthCanMessage,struct ECMSG>::createBuffer(int)" (__imp_?createBuffer@?$ImplementCanBufferFactory@VEthCanMessage@@UECMSG@@@dev@yarp@@UEAA?AVCanBuffer@23@H@Z) referenced in function "public: virtual class yarp::dev::CanBuffer __cdecl eDriver::createBuffer(int)" (?createBuffer@eDriver@@UEAA?AVCanBuffer@dev@yarp@@H@Z) [C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\FirmwareUpdater\FirmwareUpdater-qt.vcxproj]
  canLoaderLib.lib(driver.obj) : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual void __cdecl yarp::dev::ImplementCanBufferFactory<class EthCanMessage,struct ECMSG>::destroyBuffer(class yarp::dev::CanBuffer &)" (__imp_?destroyBuffer@?$ImplementCanBufferFactory@VEthCanMessage@@UECMSG@@@dev@yarp@@UEAAXAEAVCanBuffer@23@@Z) referenced in function "public: virtual void __cdecl eDriver::destroyBuffer(class yarp::dev::CanBuffer &)" (?destroyBuffer@eDriver@@UEAAXAEAVCanBuffer@dev@yarp@@@Z) [C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\FirmwareUpdater\FirmwareUpdater-qt.vcxproj]
  C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\bin\Release\FirmwareUpdater.exe : fatal error LNK1120: 2 unresolved externals [C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\FirmwareUpdater\FirmwareUpdater-qt.vcxproj]

"C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\canLoader\canLoader-console\canLoader-console.vcxproj" (default target) (13) ->
  canLoaderLib.lib(driver.obj) : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual class yarp::dev::CanBuffer __cdecl yarp::dev::ImplementCanBufferFactory<class EthCanMessage,struct ECMSG>::createBuffer(int)" (__imp_?createBuffer@?$ImplementCanBufferFactory@VEthCanMessage@@UECMSG@@@dev@yarp@@UEAA?AVCanBuffer@23@H@Z) referenced in function "public: virtual class yarp::dev::CanBuffer __cdecl eDriver::createBuffer(int)" (?createBuffer@eDriver@@UEAA?AVCanBuffer@dev@yarp@@H@Z) [C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\canLoader\canLoader-console\canLoader-console.vcxproj]
  canLoaderLib.lib(driver.obj) : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual void __cdecl yarp::dev::ImplementCanBufferFactory<class EthCanMessage,struct ECMSG>::destroyBuffer(class yarp::dev::CanBuffer &)" (__imp_?destroyBuffer@?$ImplementCanBufferFactory@VEthCanMessage@@UECMSG@@@dev@yarp@@UEAAXAEAVCanBuffer@23@@Z) referenced in function "public: virtual void __cdecl eDriver::destroyBuffer(class yarp::dev::CanBuffer &)" (?destroyBuffer@eDriver@@UEAAXAEAVCanBuffer@dev@yarp@@@Z) [C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\canLoader\canLoader-console\canLoader-console.vcxproj]
  C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\bin\Release\canLoader-console.exe : fatal error LNK1120: 2 unresolved externals [C:\Users\isorrentino\dev\robotology-superbuild\build\robotology\ICUB\src\tools\canLoader\canLoader-console\canLoader-console.vcxproj]

    0 Warning(s)
    6 Error(s)

Time Elapsed 00:00:07.12
~~~

traversaro, 2019-05-09:
Caused by a problem introduced in https://github.com/robotology/yarp/pull/2006, fixed by https://github.com/robotology/yarp/pull/2017 .

traversaro, 2019-05-09:
Fixed by https://github.com/robotology/yarp/pull/2017 .

closed 2019-05-09 by traversaro