robotology/icub-main #578: Black lines in camera feed

fbottarel, 2019-05-15:
### Description of the failure
Black lines started showing up at the top of the image streams

### Detailed conditions and logs of the failure
The Dragonfly cameras output this kind of video stream at 320x240 (image is magnified for better identification of the issue). The top rows of pixels seem corrupted. 

[image]

By resetting the camera via the frameGrabberGui, the image goes back to default parameters (640x480 resolution) and the feed corruption seems no longer present. 

[image]

Rebooting either the cameras, the pc104 or the whole robot does not solve the problem.

Nicogene, 2019-05-15:
It would be useful to know if it is reproducible also with `usbCamera` (I guess yes) and if it affects `icub-main` on `master` or `devel` branch.

Nicogene, 2019-05-15:
I tryed with latest `devel` both `icub-main` and `yarp` and the image look fine also 320x240 using the `usbCamera`.
Could be the bayer? cc @barbalberto

fbottarel, 2019-05-16:
I can get some raw image dumps and try on my laptop before updating the robot machines.

Nicogene, 2019-05-16:
Usually those artifacts happens when the header is considered has part of the image, probably there is some byte alignment issue with the Bayer coding.

fbottarel, 2019-05-16:
I thought we used Bayer encoding only for streaming 640x480 images, does that also happen for lower res?

frank-foerster, 2019-10-22:
Just to confirm, that I observed a similar or same problem. In my case the "flickering bar" is not at the very top and broader than what is depicted in the gifs above. I am using Bayer encoding here as recommended in the Wiki.
[image]

When trying to produce the gifs for this entry, once it ended up being as bad as in the following, but one one occasion no flickering bar at all occurred:
[image]

valegagge, 2026-09-28:
This issue had been fixed with this PR https://github.com/robotology/icub-main/pull/1071 and https://github.com/robotology/icub-main/pull/1072.

Tested on icubGenova11 :

[image]

closed 2026-09-28 by valegagge