robotology/icub-main #581: More control input modes in 2FOC boards.

ale-git, 2019-06-11:
The purpose of this issue is to keep track of the icub-main parts touched by the improvements in the low level motor control. The new EMS board firmware will be able to exploit features already present in the 2FOC motor board. In particular, we'll be able to control the iCub motors by velocity and current closed control loops in addition to the pwm open loop now available.

https://github.com/ale-git/icub-main/tree/flex_motor_input

@nunoguedelha

nunoguedelha, 2019-06-11:
Ciao @randaz81 , with @ale-git we started discussing about the integration of these changes and the dependencies (`icub-firmware-shared`) on the `devel` branch of the respective repos...

- By combining these changes with the robots configuration PRed in https://github.com/robotology/robots-configuration/pull/100 (and replicate the changes to iCubGenova02), there is no functional impact on the controllers running in the EMS (position, torque, etc) as they will keep driving the motors through a PWM signal, as done until now.

- We have tested this backward compatibility on iCubGenova04. As a side note, **iCubGenova04** and **iCubGenova02** are the only robots impacted by this feature (ETHERNET robots in `devel` branch).

- Regarding the current control mode, we started running tests on the iCubGenova04, which requires to have a new firmware build flashed in the EMS and 2FOC boards (that build is already merged in the `devel` of `icub-firmware-build` repo). To ease the integration tests on iCubGenova04 and avoid flashing firmware builds back and forth as different users use the robot, we wish to keep the required firmware build flashed permanently.

For that reason we need to merge as soon as possible all these dependent changes:
- `robots-configuration`
- `icub-main`
- `icub-firmware-shared`

Could we meet on Thursday in order to discuss F2F?

CC @DanielePucci @fjandrad @S-Dafarra @gabrielenava

nunoguedelha, 2019-07-10:
ciao @ale-git , as discussed initially with @randaz81, could you please add a documentation on the configuration of the controllers regarding the new "flexible motor input", i.e. the choice of the output type: PWM, Current or Velocity?
Thx.

nunoguedelha, 2019-07-11:
Just to make sure we don't miss any points discussed during the meeting, I'm posting the meeting minutes..

nunoguedelha, 2019-07-11:
# Meeting minutes (11 July 2019) - [WIP]

**Organizer:** Marco Accame

**Participants:** Marco Accame, Alessandro Scalzo, Nuno Guedelha, Marco Randazzo, Ugo Pattacini, Valentina Gaggero

**Discussion points:**

* Status of activity (ale)
* Results of tests on robots (nuno, ale)
* Decisions on how to merge this feature into devel and when to do it (all)

**Short overview of the feature:**

Previous to the implementation of the "flexible motor input"(*) by @ale-git, each controller (Position, Position Direct, Torque, Velocity) running in the EMS board @1KHz was generating the same type of output, a PWM setpoint, from the desired respective position, torque or velocity and respective feed-forward and PID parameters. That output was then fed to the motor driver running in the 2FOC @20khz, that would modulate the PWM on each motor phase according to the rotor position.
After the implementation of the "flexible motor input", each controller can output a desired PWM, current or velocity that is fed to the motor driver (2FOC), where it is converted to the ultimate PWM setpoint. In the case of a desired current input, that driver closes the loop on the measured current (@20khz).

## Status of the activity and results of tests on the robots

### List of changes impacting the multiple robot software components

- `icub-main`: branch `flex_motor_input` ⟹ https://github.com/robotology/icub-main/issues/581
- `icub-firmware-shared`: branch `fmi` ⟹ https://github.com/robotology/icub-firmware-shared/issues/28
- `icub-firmware`: `flexible_motor_input` ⟹ https://github.com/robotology/icub-firmware/issues/86
- `icub-firmware-build`: branch `devel`, commit `82f9b234dbb19c78100d2eb13e1064e726eb800c`
- `robots-configuration`: branch `feature/rebase-curent-control-cfg` (`hardware/motorcontrol` files for specifying the controllers output type PWM, velocity or current) ⟹ https://github.com/robotology/robots-configuration/pull/100
(backward compatibility with the position and torque control using PWM output: this change allows to run the new firmware and `icub-main` dependencies on the robot without impacting the present behaviour of the position and torque low level controllers, i.e. using the PWM output of these controllers, and not the new current nor velocity output.)

The backward compatibility has been tested on iCubGenova04 (Green iCub), running basic motions and complex dynamic demos like the Yoga and the Walking (DIC team)

Main entry point issue tracking the tests on iCubGenova04 and the test bench (iCub V2.5 and V3 motors):
https://github.com/dic-iit/element_torque-control-via-current/issues/56

### List of issues

Main issues:
* [Motor current offset varies with the rotor position (iCubV2.5)](https://github.com/dic-iit/element_torque-control-via-current/issues/52), [Motor current offset varies with the rotor position (iCubV3)](https://github.com/dic-iit/element_torque-control-via-current/issues/59):
    - We observe oscillations in the quadrature current, depending on the rotor position.
    - We observed phase small angle errors (6 deg in the electrical phase space) in the phase shift between phase currents and phase PWMs.
    - This has been observed while testing the iCub V2.5 and the iCub V3 single motors on a dedicated test bench, after opening the loops that track the direct and quadrature currents (all PID gains set to 0).
    - @ale-git suspects this to be caused by rounding errors in the Clark-Park transform computations.

* [Implement and test the PWM modulation from the motor encoder only](https://github.com/dic-iit/element_torque-control-via-current/issues/67)

Tests are currently ongoing for verifying if the improvements implemented for the [second issue](https://github.com/dic-iit/element_torque-control-via-current/issues/67), while fixing rounding errors, also fix the [motor quadrature current oscillations issue](https://github.com/dic-iit/element_torque-control-via-current/issues/59).

## Decisions on how to merge this feature into development and when to do it:

We discussed on the choice between the following alternatives:
- propagating immediatly the changes in the PR https://github.com/robotology/robots-configuration/pull/100 to all robots configuration files
- changing the parser of the configuration files, residing in `icub-main`, for being backward compatible with the old configuration format.

Although all the configuration files should converge in the long term (as mentioned by Hugo), most probably by applying the automatic propagation/update script by @randaz81, the 2nd alternative was prefered: making the `icub-main` parser compatible with the old configuration format (TO BE CONFIRMED BY @marcoaccame AND @ale-git).

### Next Actions
- [ ] Run tests of backward compatible `flexible motor input` feature on iCubGenova02 (Green iCub) -> Nuno G.
- [ ] Run tests of backward compatible `flexible motor input` feature on R1 robots -> Marco R.
- [ ] Push to PR https://github.com/robotology/robots-configuration/pull/100 the update on the configuration [templates](https://github.com/robotology/robots-configuration/tree/devel/iCubTemplates) -> Alessandro S.
- [ ] Update the robots configuration parser in `icub-main` for backward compatibility: set missing parameters (`velocityControl`, `mixedControl`, `mixedControl`, `outputType`) to default values, and translate the `controlLaw` field to the respective new values -> Alessandro S. (TO BE CONFIRMED BY @marcoaccame AND @ale-git)

CC @marcoaccame @valegagge  @ale-git @randaz81 @pattacini

nunoguedelha, 2019-07-11:
CC @DanielePucci

DanielePucci, 2019-07-11:
Great guys, thanks a lot. Just remember to take a [system perspective](https://necsi.edu/system-perspective) during this discussion for implementing nonlinearities and feedforward compensation at the firmaware level

nunoguedelha, 2019-07-16:
ciao @ale-git , have you finished rebasing your branches on `icub-main`, `icub-firmware` and `icub-firmware-shared` so that I can run the tests on the purple iCub this afternoon? Thx

ale-git, 2019-07-16:
Hi @nunoguedelha , I finished right now. I've put the ems004fmi.hex and mc4plusfmi.hex files of EMS and MC4+ boards in icub-firmware-build/testing.

nunoguedelha, 2019-07-17:
Ciao @ale-git , I've tested your updated branches (`icub-main`, `icub-firmware-shared`) and firmware builds, along with the robot configuration files (https://github.com/robotology/robots-configuration/pull/100/commits), on the purple iCub (`iCubGenova02`), and everything went fine. More details [here](https://github.com/robotology/icub-support/issues/843).

nunoguedelha, 2019-07-17:
Ciao @ale-git , @randaz81 ,
any update on the pending actions?

Regarding the next meeting I suggest we do it next Monday. What do you think?
@marcoaccame @valegagge @ale-git @randaz81 @pattacini

ale-git, 2019-07-17:
Hi @nunoguedelha I'm completing the parser back compatibility, I think that I'll be able to test it tomorrow. About the meeting I suggest Tuesday because I'm not sure I'll be in IIT on Monday.

randaz81, 2019-07-19:
Next week I'll be available  only on 22 and 26

nunoguedelha, 2019-07-19:
I think it's ok to do the meeting later as long as we merge the code on Monday before @randaz81 leaves. This implies:
- @ale-git completing the task on the parser today, such that @randaz81 can check before leaving
- @randaz81 completing R1 tests by Monday and merging the code.

If there is any obstacle preventing to achieve the plan, or if I can be of any help please let me know.

nunoguedelha, 2019-07-20:
https://github.com/robotology/cer/pull/105#issuecomment-513441969
> @randaz81 @nunoguedelha the latest commit (back compatible version) is still under debugging. I'll be in IIT on monday btw. Please use version [ale-git/icub-main@6cf4ade](https://github.com/ale-git/icub-main/commit/6cf4ade9be8f2cf05ee2edd6736f22d9cdaeb623) (the same used by Nuno on icub robots) to test the new configuration files. Alternatively, I can test the back compatible version on R1 too.

nunoguedelha, 2019-07-30:
These changes were implemented in #585
The analysis and implementation of the backward compatible parser will be tracked in https://github.com/robotology/icub-main/issues/593 for the following reasons:
- the code implementing the new "flex motor input" is already merged.
- the priority of the backward compatible parser has been decreased since it is required only by the merging of `devel` into `master`.
- the full update of all the `robots-configuration` files could eventually be ready soon enough to make the backward compatible parser unnecessary.

I suggest we close this issue.

nunoguedelha, 2019-07-30:
CC @marcoaccame

pattacini, 2021-03-27:
Closing, feel free to open it up again if needed.

closed 2021-03-27 by pattacini