robotology/icub-main #593: Implement a robots-configuration parser backward compatible with the configuration format used before #581

nunoguedelha, 2019-07-30:
Implement a robots-configuration parser backward compatible with the configuration format used before the changes introduced by #581. This way, the missing parameters required by the new low level motor controller interface will be set automatically when parsing old formatted configuration files. For more details refer to https://github.com/robotology/icub-main/issues/581#issuecomment-510576358.

- [ ] Update the robots configuration parser in `icub-main` for backward compatibility: set missing parameters (`velocityControl`, `mixedControl`, `mixedControl`, `outputType`) to default values, and translate the `controlLaw` field to the respective new values.
- [ ] Update the `robots-configuration` [templates](https://github.com/robotology/robots-configuration/tree/devel/iCubTemplates).

nunoguedelha, 2019-07-30:
@ale-git @marcoaccame this is just for tracking the already ongoing work by @ale-git .

pattacini, 2021-03-27:
Closing, feel free to open it up again if needed.

closed 2021-03-27 by pattacini