robotology/icub-main #597: Robot interface startup FAILS after update of EMS, MC4PLUS and 2FOC builds to the latest devel

nunoguedelha, 2019-09-02:
After updating the firmware EMS, MC4PLUS and 2FOC builds to the latest devel commit https://github.com/robotology/icub-firmware-build/commit/46d3f477b566be0b01d34409987e5d259abe36bb, the Robot interface startup fails. There is no ERROR message, only WARNINGS like the following:
```
3832	56563.465063	WARNING	 from BOARD 10.0.1.25 (left_arm-eb25-j8_11) @ 1908s 727m 922u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 1 eobrd_mais boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
3864	56563.688643	WARNING	 from BOARD 10.0.1.26 (left_arm-eb26-j12_15) @ 1908s 957m 154u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 1 eobrd_mais boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
3971	56564.450389	WARNING	 from BOARD 10.0.1.28 (right_arm-eb28-j8_11) @ 1909s 707m 168u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 1 eobrd_mais boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
4002	56564.673772	WARNING	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15) @ 1909s 928m 396u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 1 eobrd_mais boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
4295	56567.250317	WARNING	 from BOARD 10.0.1.24 (left_arm-eb24-j4_7) @ 1912s 512m 732u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 7 eobrd_mtb boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
4317	56568.103766	WARNING	 from BOARD 10.0.1.27 (right_arm-eb27-j4_7) @ 1913s 362m 870u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 7 eobrd_mtb boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
4358	56568.959286	WARNING	 from BOARD 10.0.1.11 (right_leg-eb11-skin) @ 1913s 984m 273u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 13 eobrd_mtb boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
4390	56569.570182	WARNING	 from BOARD 10.0.1.22 (face-eb22-j0_1) @ 1914s 832m 45u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 4 eobrd_mtb boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
```
These are with respect to the MC4PLUS boards startup. For instance, for the MC4PLUS at IP 29 we get the following sequence of Warning/Info messages:
```
3992	56564.633390	WARNING	embObjMC BOARD  right_arm-eb29-j12_15 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails)
3994	56564.633415	INFO	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 1909s 892m 618u: (code 0x00000034, par16 0x0001 par64 0x0000000000000023) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
4000	56564.663659	INFO	EthResource::askBoardVersion() found BOARD right_arm-eb29-j12_15 @ IP 10.0.1.29 of type mc4plus with FW = ver 3.21 built on 2019 Jul 24 17:34
4001	56564.673752	INFO	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15) @ 1909s 928m 234u: CAN discovery has started for 1 eobrd_mais boards on (can1map, can2map) = (0x4000, 0x0000) with target can protocol ver 0.0 and application ver 0.0.0.
4002	56564.673772	WARNING	 from BOARD 10.0.1.29 (right_arm-eb29-j12_15) @ 1909s 928m 396u: CAN discovery is OK but FAKE (without any control on CAN w/ get-fw-version<> message) for 1 eobrd_mais boards with target can protocol ver 0.0 and application ver 0.0.0. Search time was 0 ms
```
Something goes wrong with the CAN discovery, and/or there is some missing configuration parameter.

The respective boards are detected properly through `FirmwareUpdater`.

traversaro, 2019-09-02:
@nunoguedelha done

valegagge, 2019-09-02:
Hi @nunoguedelha ,
the `enableTemperatureTransmission` function is private, so none can calls it. It is used inside to enable and disable the transmission of temperature values during opening or closing device.

The idea behind the return value is: return true if the command has been sent, false otherwise. If the resource doesn't exist then it is not possible sent the command. This happen only:
1) if the resource has been close twice
2) if the device has been created, but never opened.

In both case, at the end we are sure that the transmission is disable.
Anyway, I think is not simple find the correct trade off between provide information to the user and to make the log readable. Therefore if you prefer to add a warning message, feel free to change the file. :)
 
About the second case you mentioned, it is not possible because the  `enableTemperatureTransmission` is called after the creation of resource.

Instead, I see that the return value is not check in open function, so I verify it [here](https://github.com/robotology/icub-main/commit/55154efbb1d94983598f85e270c6ad145a3ba4de)

nunoguedelha, 2019-09-02:
Hi @valegagge , it's clear thanks.

I was thinking also about a safeguard mechanism for the case where `enableTemperatureTransmission` is called from inside the device `embObjFTsensor` before the resource is allocated, by mistake (after some code refactoring for instance). But this is just my personal point of view regarding protection against human coding errors.

nunoguedelha, 2019-09-02:
I will check tomorrow the fix.

nunoguedelha, 2019-09-02:
Updated comment https://github.com/robotology/icub-main/issues/597#issuecomment-527147375

nunoguedelha, 2019-09-10:
@fjandrad tested the fix and it was working fine.
Both fixes by @valegagge are commits bb055cd and 55154ef.

Current branch `devel` of the repo is already ahead of these commits.

Closing the issue.

closed 2019-09-10 by nunoguedelha