robotology/icub-main #601: Unable to find yarp/dev/PreciselyTimed.h

GiulioRomualdi, 2019-09-20:
After this commit in YARP https://github.com/robotology/yarp/commit/e080845de607bbd264791c179a84babac9a650eb `icub-main` was broken

This is the error
```
/home/gromualdi/robot-code/robotology-superbuild/robotology/ICUB/src/tools/controlBoardDumper/genericControlBoardDumper.h:23:10: fatal error: yarp/dev/PreciselyTimed.h: No such file or directory
 #include <yarp/dev/PreciselyTimed.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
The bug can be easily fixed by with the following patch in [`genericControlBoardDumper.h`](https://github.com/robotology/icub-main/blob/master/src/tools/controlBoardDumper/genericControlBoardDumper.h)
```patch
--- #include <yarp/dev/PreciselyTimed.h>
+++ #include <yarp/dev/IPreciselyTimed.h>
```

GiulioRomualdi, 2019-09-20:
If you want I can open a PR with the fix

pattacini, 2019-09-20:
Fixed in @d188e15
Thanks @GiulioRomualdi for reporting.

commit d188e151 (2019-09-20): Update genericControlBoardDumper.h

fix #601
closed 2019-09-20 by pattacini