robotology/icub-main #604: icub-main build (with icub-head configuration) failing because of missing dependency on HarBuzz

nunoguedelha, 2019-09-26:
## Issue description
When building `icub-main` with a configuration set for the icub-head, we build the target `canLoader` which fails because a dependency cannot be found (header file `hb.h` -> HarfBuzz):

Issue occurs on MacOSX Sierra.

## Build log
```
=== BUILD TARGET canLoader OF PROJECT iCub WITH CONFIGURATION Debug ===

Check dependencies

Write auxiliary files
/bin/mkdir -p /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/Objects-normal/x86_64
write-file /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/Objects-normal/x86_64/canLoader.LinkFileList

CompileC /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/Objects-normal/x86_64/calibrate_window.o src/tools/canLoader/canLoader/calibrate_window.cpp normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/nunoguedelha/dev/robotology-superbuild/robotology/ICUB
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c++ -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -Wno-trigraphs -fpascal-strings -O0 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-float-conversion -Wno-non-literal-null-conversion -Wno-objc-literal-conversion -Wno-shorten-64-to-32 -Wno-newline-eof -Wno-c++11-extensions -DCMAKE_INTDIR=\"Debug\" -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Winvalid-offsetof -mmacosx-version-min=10.12 -g -Wno-sign-conversion -Wno-infinite-recursion -Wno-move -Wno-comma -Wno-block-capture-autoreleasing -Wno-strict-prototypes -Wno-range-loop-analysis -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/bin/Debug/include -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/ICUB/src/tools/canLoader/canLoader/../canLoaderLib -I/Users/nunoguedelha/dev/robotology-superbuild/build/install/include/icub-firmware-shared -I/Users/nunoguedelha/dev/robotology-superbuild/build/install/include/icub-firmware-shared/canProtocolLib -I/Users/nunoguedelha/dev/robotology-superbuild/build/install/include/icub-firmware-shared/embobj/plus/comm-v2/icub -I/Users/nunoguedelha/dev/robotology-superbuild/build/install/include/icub-firmware-shared/embobj/core/core -isystem /usr/local/include/freetype2 -isystem /usr/local/include -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/System/Library/Frameworks/Tk.framework/Headers -isystem /usr/local/include/glib-2.0 -isystem /usr/local/lib/glib-2.0/include -isystem /usr/local/include/atk-1.0 -isystem /usr/local/include/gdk-pixbuf-2.0 -isystem /usr/local/include/cairo -isystem /usr/local/include/pango-1.0 -isystem /usr/local/include/gtk-2.0 -isystem /usr/local/lib/gtk-2.0/include -isystem /usr/local/include/sigc++-2.0 -isystem /usr/local/lib/sigc++-2.0/include -isystem /usr/local/include/glibmm-2.4 -isystem /usr/local/lib/glibmm-2.4/include -isystem /usr/local/include/giomm-2.4 -isystem /usr/local/lib/giomm-2.4/include -isystem /usr/local/include/atkmm-1.6 -isystem /usr/local/include/cairomm-1.0 -isystem /usr/local/lib/cairomm-1.0/include -isystem /usr/local/include/pangomm-1.4 -isystem /usr/local/lib/pangomm-1.4/include -isystem /usr/local/include/gdkmm-2.4 -isystem /usr/local/lib/gdkmm-2.4/include -isystem /usr/local/include/gtkmm-2.4 -isystem /usr/local/lib/gtkmm-2.4/include -isystem /Users/nunoguedelha/dev/robotology-superbuild/build/install/include -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/DerivedSources/x86_64 -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/DerivedSources -Wmost -Wno-four-char-constants -Wno-unknown-pragmas -F/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/bin/Debug -fPIE -std=c++11 -MMD -MT dependencies -MF /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/Objects-normal/x86_64/calibrate_window.d --serialize-diagnostics /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/Objects-normal/x86_64/calibrate_window.dia -c /Users/nunoguedelha/dev/robotology-superbuild/robotology/ICUB/src/tools/canLoader/canLoader/calibrate_window.cpp -o /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/ICUB/src/tools/canLoader/canLoader/iCub.build/Debug/canLoader.build/Objects-normal/x86_64/calibrate_window.o
In file included from /Users/nunoguedelha/dev/robotology-superbuild/robotology/ICUB/src/tools/canLoader/canLoader/calibrate_window.cpp:10:
/Users/nunoguedelha/dev/robotology-superbuild/robotology/ICUB/src/tools/canLoader/canLoader/../canLoaderLib/downloader.h:245:24: warning: field 'calibrated' will be initialized after field 'valid' [-Wreorder]
    strain_value_t() : calibrated(false), valid(false), saturated(false) {
                       ^
In file included from /Users/nunoguedelha/dev/robotology-superbuild/robotology/ICUB/src/tools/canLoader/canLoader/calibrate_window.cpp:11:
In file included from /Users/nunoguedelha/dev/robotology-superbuild/robotology/ICUB/src/tools/canLoader/canLoader/calibrate_window.h:13:
In file included from /usr/local/include/gtk-2.0/gtk/gtk.h:32:
In file included from /usr/local/include/gtk-2.0/gdk/gdk.h:32:
In file included from /usr/local/include/gtk-2.0/gdk/gdkapplaunchcontext.h:31:
In file included from /usr/local/include/gtk-2.0/gdk/gdkscreen.h:32:
In file included from /usr/local/include/gtk-2.0/gdk/gdktypes.h:37:
In file included from /usr/local/include/pango-1.0/pango/pango.h:25:
In file included from /usr/local/include/pango-1.0/pango/pango-attributes.h:25:
In file included from /usr/local/include/pango-1.0/pango/pango-font.h:25:
/usr/local/include/pango-1.0/pango/pango-coverage.h:28:10: fatal error: 'hb.h' file not found
#include <hb.h>
         ^~~~~~
1 warning and 1 error generated.
```

Pango (https://pango.gnome.org/) dependency is used for handling the GTK2 font. Pango depends on [HarfBuzz](https://pango.gnome.org/HarfBuzz).

That dependency is installed pon my System, but the include path is missing in the cmake configuration.

nunoguedelha, 2019-09-26:
@marcoaccame or @valegagge could you give me a hand on this one?

traversaro, 2019-09-27:
Hi @nunoguedelha, the problem is probably in the CMake files for supporting GTK2 and HarfBuzz. However, the GTK2-based GUIs have been deprecated for a long time (see https://github.com/robotology/icub-main/issues/193), and so it is perfectly fine to just disable their compilation, by setting the `ICUB_USE_GTK2` CMake cache variable of the iCub build to `OFF` . The reason why it was set to `ON` on your system is that all the `ICUB_USE_<dep>` variables inspect the system, and if a given dependency `<dep>` is found, they are set to `ON` automatically (see https://github.com/robotology/icub-main/blob/master/conf/iCubFindDependencies.cmake#L7). This behavior makes sense for tested dependencies, but clearly is not a good idea for deprecated and not tested dependencies, however this part of code will soon be removed, so this problem will be solved by that.

nunoguedelha, 2019-09-27:
Thanks @traversaro ! closing the issue.

closed 2019-09-27 by nunoguedelha