robotology/icub-main #606: Remove gkt-based modules

traversaro, 2019-09-27:
Sub-issue of https://github.com/robotology/icub-main/issues/193 .

traversaro, 2019-09-27:
@pattacini mentioned that it could be a good idea to move the code (even in a state that is not compiling) in https://github.com/robotology-legacy .

traversaro, 2019-09-27:
As a preliminary step, we explicitly set `ICUB_USE_GTK2` to `OFF` in the robotology-superbuild in https://github.com/robotology/robotology-superbuild/pull/266 .

Nicogene, 2019-09-27:
> @pattacini mentioned that it could be a good idea to move the code (even in a state that is not compiling) in https://github.com/robotology-legacy .

I agree

pattacini, 2019-09-27:
For porting the code, a simple fork in the dedicated legacy org should be sufficient.

pattacini, 2020-05-15:
Done in #659 by @Nicogene ✨

PR #659 (2020-05-15): Remove legacy gtk modules
This PR get rid of the legacy gtk-based modules.
They have been moved here: https://github.com/robotology-legacy/icub-gtk-modules

Here is the list of the removed modules:
- `canLoader`
- `ethLoader`
- `focus-tool`
- `iCubSkinGui-gtk`
- `skinManagerGui-gtk`
- `strainCalibGUI`
- `strainCalibrationDataAcquisition`

In particular I need to know if `canLoader`, `ethLoader`, `focus-tools`, `strainCalibGUI`, `strainCalibrationDataAcquisition` can be dismissed.

It fixes #606
closed 2020-05-15 by pattacini