robotology/icub-main #611: camCalibWithPose crashes on Windows

GiulioRomualdi, 2019-10-08:
I am trying to use `camCalibWithPose`  on a Windows machine and I have the following problems:
1.  When I tried to connect the ports (`\icub\head\state:o` and the `icub\cam\left`) the program crashed. We (@kouroshD) compiled the code in `RelWithDebugInfo` and we discovered that it crashed at the following line https://github.com/robotology/icub-main/blob/0a2e1d909fd1466565b2bcab4f755d59e61f7451/src/modules/camCalibWithPose/src/CamCalibModule.cpp#L110
   With the following error 
    ```
    Unhandled exception thrown: read access violation. pixel was 0x181FCEE3000
    ```
    [image]

2. camCalibWithPose sometimes does not crashes but `icub/camcalib/left/out/` is always empty

Do you have any idea? @drdanz @S-Dafarra @kouroshD @randaz81

pattacini, 2019-11-15:
@Nicogene [this commit][1] removed from the triplet {`camCalib`, `camCalibWithPose`, `dualCamCalib`} the memory allocation of the image `out`. Can this be potentially causing the crash as the following call comes right before the line @GiulioRomualdi has spotted?
https://github.com/robotology/icub-main/blob/0a2e1d909fd1466565b2bcab4f755d59e61f7451/src/modules/camCalibWithPose/src/CamCalibModule.cpp#L105

Don't know why `camCalib` seems to be working though...

[1]: https://github.com/robotology/icub-main/commit/7a235125cf9d188270c29ce32cdc6e200d813276#diff-15887b0326c3d2d40911667ef7299ea4

Nicogene, 2019-11-15:
Actually all the camera calibration software have problems since a while (e.g. https://github.com/robotology/icub-main/issues/478). 
This segfault on windows happens systematically?
I used `camCalibWithPose` with @drdanz for the oculus and it worked on windows but I don't know honestly if it was the last version of `icub-main`.

Nicogene, 2019-11-15:
Another question

>When I tried to connect the ports (\icub\head\state:o and the icub\cam\left) the program crashed. We (@kouroshD) compiled the code in RelWithDebugInfo and we discovered that it crashed at the following line

I don't know how `camCalibWIthPose` specifically works, but why are you connecting the state of the head and the port of the camera??

traversaro, 2019-11-15:
If you need to debug memory problems on Windows, Dr. Memory is a great tool even if the homepage is really '90 ( https://drmemory.org/ ).

Nicogene, 2019-11-15:
Chatting with @GiulioRomualdi it emerged that it happens systematically and that it happens **only** on windows. On Linux it works fine.

pattacini, 2019-12-11:
Hi @GiulioRomualdi 

In the end, we judged that `camCalibWithPose` needs a deep refactoring, thus we haven't planned any debug activity to devote to this, in its present version.

For the task of refactoring, @Nicogene has opened up a dedicated issue on our internal workflow (https://github.com/icub-tech-iit/fix/issues/133).

closed 2019-12-11 by pattacini