robotology/icub-main #619: Installation fails due to stereoCalib on Ubuntu 18.04

prajval10, 2019-11-08:
I followed the instructions from http://wiki.icub.org/wiki/Linux:Installation_from_sources#Getting_all_dependencies and the cmake options specified in the docs. I have errors as seen here:

```[console]
[ 52%] Building CXX object src/tools/stereoCalib/CMakeFiles/stereoCalib.di/src/stereoCalibThread.cpp.o
In file included from /home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv.h:60:0,
                 from /home/pmurali/icub-main/src/tools/stereoCalib/src/stereoCalibThread.cpp:2:
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:45:89: error: ‘CV_RGB2BGR’ was not declared in this scope
 template<> struct convert_code_to_cv<yarp::sig::PixelRgb> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                         ^~~~~~~~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:45:89: note: suggested alternative: ‘CV_RGB’
 template<> struct convert_code_to_cv<yarp::sig::PixelRgb> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                         ^~~~~~~~~~
                                                                                         CV_RGB
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:45:99: error: template argument 2 is invalid
 template<> struct convert_code_to_cv<yarp::sig::PixelRgb> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                                   ^
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:46:95: error: ‘CV_RGB2BGR’ was not declared in this scope
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbSigned> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                               ^~~~~~~~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:46:95: note: suggested alternative: ‘CV_RGB’
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbSigned> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                               ^~~~~~~~~~
                                                                                               CV_RGB
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:46:105: error: template argument 2 is invalid
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbSigned> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                                         ^
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:47:94: error: ‘CV_RGB2BGR’ was not declared in this scope
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbFloat> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                              ^~~~~~~~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:47:94: note: suggested alternative: ‘CV_RGB’
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbFloat> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                              ^~~~~~~~~~
                                                                                              CV_RGB
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:47:104: error: template argument 2 is invalid
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbFloat> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                                        ^
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:48:92: error: ‘CV_RGB2BGR’ was not declared in this scope
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbInt> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                            ^~~~~~~~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:48:92: note: suggested alternative: ‘CV_RGB’
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbInt> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                            ^~~~~~~~~~
                                                                                            CV_RGB
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:48:102: error: template argument 2 is invalid
 template<> struct convert_code_to_cv<yarp::sig::PixelRgbInt> : std::integral_constant<int, CV_RGB2BGR> {};
                                                                                                      ^
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:49:89: error: ‘CV_HSV2BGR’ was not declared in this scope
 template<> struct convert_code_to_cv<yarp::sig::PixelHsv> : std::integral_constant<int, CV_HSV2BGR> {};
                                                                                         ^~~~~~~~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:49:89: note: suggested alternative: ‘CV_SSE2’
 template<> struct convert_code_to_cv<yarp::sig::PixelHsv> : std::integral_constant<int, CV_HSV2BGR> {};
                                                                                         ^~~~~~~~~~
                                                                                         CV_SSE2
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:49:99: error: template argument 2 is invalid
 template<> struct convert_code_to_cv<yarp::sig::PixelHsv> : std::integral_constant<int, CV_HSV2BGR> {};
                                                                                                   ^
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:50:94: error: ‘CV_HSV2BGR’ was not declared in this scope
 template<> struct convert_code_to_cv<yarp::sig::PixelHsvFloat> : std::integral_constant<int, CV_HSV2BGR> {};
                                                                                              ^~~~~~~~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:50:94: note: suggested alternative: ‘CV_SSE2’
 template<> struct convert_code_to_cv<yarp::sig::PixelHsvFloat> : std::integral_constant<int, CV_HSV2BGR> {};
                                                                                              ^~~~~~~~~~
                                                                                              CV_SSE2
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:50:104: error: template argument 2 is invalid
 template<> struct convert_code_to_cv<yarp::sig::PixelHsvFloat> : std::integral_constant<int, CV_HSV2BGR> {};
                                                                                                        ^
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:51:90: error: ‘CV_RGBA2BGRA’ was not declared in this scope
 template<> struct convert_code_to_cv<yarp::sig::PixelRgba> : std::integral_constant<int, CV_RGBA2BGRA> {};
                                                                                          ^~~~~~~~~~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:51:90: note: suggested alternative: ‘CV_RGB’
[... 166 lines cut ...]
                 from /home/pmurali/icub-main/src/tools/stereoCalib/src/stereoCalibThread.cpp:2:
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h: In instantiation of ‘cv::Mat yarp::cv::toCvMat(yarp::sig::ImageOf<T>&) [with T = yarp::sig::PixelRgb]’:
/home/pmurali/icub-main/src/tools/stereoCalib/src/stereoCalibThread.cpp:180:47:   required from here
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:82:38: error: ‘value’ is not a member of ‘yarp::cv::convert_code_to_cv<yarp::sig::PixelRgb>’
     if (convert_code_to_cv<T>::value >= 0)
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
/home/pmurali/yarp/src/libYARP_cv/include/yarp/cv/Cv-inl.h:84:23: error: ‘value’ is not a member of ‘yarp::cv::convert_code_to_cv<yarp::sig::PixelRgb>’
         ::cv::cvtColor(outMat, outMat, convert_code_to_cv<T>::value);
         ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/build.make:110: recipe for target 'src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/src/stereoCalibThread.cpp.o' failed
make[2]: *** [src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/src/stereoCalibThread.cpp.o] Error 1
[... cut ...] arget 'src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/all' failed
make[1]: *** [src/tools/stereoCalib/CMakeFiles/stereoCalib.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

However build is fine with
```
 ICUB_USE_OpenCV                  OFF                                                                                                                                 
 ICUB_USE_OpenGL                  OFF 
```
My OpenCV version: 4.1.0

CC: @traversaro

traversaro, 2019-11-08:
Hi Praj, unfortunatly the fact that OpenCV 4 is not supported on YARP/ICUB and the related stack is a known issue, see https://github.com/robotology/yarp/issues/1672 . If you need to compile in a system with OpenCV, I suggest to disable the `ICUB_USE_OpenCV`.

traversaro, 2019-11-08:
As the issue is tracked in https://github.com/robotology/yarp/issues/1672 , I will close the issue, thanks in any case for reporting.

closed 2019-11-08 by traversaro

prajval10, 2019-11-08:
> Hi Praj, unfortunatly the fact that OpenCV 4 is not supported on YARP/ICUB and the related stack is a known issue, see [robotology/yarp#1672](https://github.com/robotology/yarp/issues/1672) . If you need to compile in a system with OpenCV, I suggest to disable the `ICUB_USE_OpenCV`.

Got it. Thanks! :+1: